robotology/community #222: gazebo-yarp | videotexture plugin loaded twice

charleswilmot, 2017-05-11:
Hello all,

I'm experiencing a problem with the video texture plugin.
Alone, the plugin is working fine. Issue arise when trying to use the plugin together with a model of the iCub in Gazebo. As explained in [this](http://answers.gazebosim.org/question/15939/visual-plugin-loaded-twice/?answer=15943#post-id-15943) thread, the plugin is loaded once on the gzserver and once in the gzclient, resulting in conflict with port access.

Is there a way to use the plugin, camera sensors and the GUI at the same time?

Best regards

traversaro, 2017-05-11:
Probably @aerydna is the best person to answer this.

charleswilmot, 2017-05-11:
I kind of hacked the code to make it work. This is how it looks like:

```
...
    m_texName      = "/"+m_model->GetName();
    if (yarp::os::NetworkBase::exists(m_texName))
    {
        yInfo()<<"[charles] "+m_texName+" already exists !!!";
        m_texName = m_texName + "_2";
    }

    yInfo()<<"[charles] opening "+m_texName;
    m_VideoPort.open(m_texName);
...
```

pattacini, 2017-05-19:
Ping @aerydna 🔔

aerydna, 2017-05-22:
hi @charleswilmot,
i can't reproduce the problem here. can you please give me all the details about the issue? which icub model are you loading in gazebo? the video texture plugin is loaded via the examble in the tutorial or in a custom sdf? thanks in advance..

pattacini, 2017-05-29:
Closing for now; feel free to add more info if needed.

charleswilmot, 2017-05-30:
Hello @aerydna ,

And sorry for the delay. Here is a screenshot of the problem:

[image]

I tried to display white noise on a screen. As you can see, the so called 'Figure 1' recieved from the robot's camera shows the white noise, whereas the GUI displays uninitialized graphic memory area. The reason is that the plugin is loaded once by the server and once by the client. It tries to open two times the same address, resulting in a conflict.

Here is the custom sdf file I'm using for the screen:
(only difference is that I removed the 'static' part)

```
<?xml version='1.0'?>
<sdf version="1.4">
<model name="my_name">
  <pose>2 0 2 1.57 0 -1.57</pose>
  <!--static>true</static-->
    <link name="my_name_link">
      <pose>0 0 0 0 0 0</pose>
      <visual name="my_name_visual">
      <geometry>
        <plane>
            <size>1 1</size>
            <normal>0 0 1</normal>
        </plane>
      </geometry>
      <material>
      <diffuse>1.0 1.0 1.0 1.0</diffuse>
      </material>
      <plugin name="VideoTexture" filename="libgazebo_yarp_videotexture.so">
      <yarpConfigurationFile>file:///tmp/screen/vtcfg.ini</yarpConfigurationFile>
      </plugin>
      </visual>
      <gravity>0</gravity>
    </link>
  </model>
</sdf>
```

The icub model I'm using is the one I found on the Robotology GitHub:
[icub_fixed](https://github.com/robotology/icub-gazebo)

I'm using Gazebo 7.7.0

Best,
Charles

charleswilmot, 2017-05-30:
Here is a simple work-around that worked for me:

VideoTexture.cc
```
/*
 * Copyright (C) 2016 Fondazione Istituto Italiano di Tecnologia & iCub Facility
 * Authors: Marco Randazzo <marco.randazzo@iit.it>
 * CopyPolicy: Released under the terms of the LGPLv2.1 or any later version, see LGPL.TXT or LGPL3.TXT
 */

#include <GazeboYarpPlugins/Handler.hh>
#include <GazeboYarpPlugins/common.h>
#include <GazeboYarpPlugins/ConfHelpers.hh>

#include <gazebo/sensors/CameraSensor.hh>
#include <VideoTexture.hh>
#include <yarp/dev/PolyDriver.h>
#include <yarp/os/Log.h>
#include <yarp/os/LogStream.h>

using namespace Ogre;
namespace gazebo
{
  VideoTexture::VideoTexture()
  {
    m_network = 0;
  }

  VideoTexture::~VideoTexture()
  {
      m_connection.reset();
      if (m_network)
      {
        delete m_network;
        m_network = 0;
      }
      m_VideoPort.close();
      if(m_material.get())
      {
        m_material->getTechnique(0)->getPass(0)->getTextureUnitState(0)->setBlank();
      }
      Ogre::TextureManager::getSingleton().remove(m_texture->getName());
  }

    void VideoTexture::Update()
    {
        ImageType* img = m_VideoPort.read(false);
        if(img)
        {
            typedef Ogre::HardwarePixelBufferSharedPtr pixelBuff;

            cv::Mat           image, converted_image;
            const cv::Mat*    image_ptr;
            pixelBuff         pixelBuffer;

            image       = cv::cvarrToMat( static_cast<IplImage*>(img->getIplImage()) );
            image_ptr   = &image;
            pixelBuffer = this->m_texture->getBuffer();

            if (image_ptr->rows != m_height || image_ptr->cols != m_width)
            {
               cv::resize(*image_ptr, converted_image, cv::Size(m_width, m_height));
               image_ptr = &converted_image;
            }

            PixelBox pb = PixelBox(m_width, m_height, 1, FORMAT, (void*)image_ptr->data);
            m_texture->getBuffer()->blitFromMemory(pb);
        }
    }

  void VideoTexture::Load(rendering::VisualPtr parent, sdf::ElementPtr sdf)
  {
    if (m_network != 0) return;

    m_network = new yarp::os::Network();
    m_model   = parent;

    if (!yarp::os::Network::checkNetwork(GazeboYarpPlugins::yarpNetworkInitializationTimeout))
    {
       yError() << "VideoTexture::Load error: yarp network does not seem to be available, is the yarpserver running?";
       return;
    }

    //Getting .ini configuration file from sdf
    bool configuration_loaded = false;

    if(!sdf->HasElement("yarpConfigurationFile") && sdf->HasElement("sdf"))
    {
        sdf = sdf->GetElement("sdf");
    }

    if (sdf->HasElement("yarpConfigurationFile"))
    {
        std::string ini_file_name = sdf->Get<std::string>("yarpConfigurationFile");
        std::string ini_file_path = gazebo::common::SystemPaths::Instance()->FindFileURI(ini_file_name);

        if (ini_file_path != "" && m_parameters.fromConfigFile(ini_file_path.c_str()))
        {
            //yInfo() << "Found yarpConfigurationFile: loading from " << ini_file_path ;
            configuration_loaded = true;
        }
     }

    if (!configuration_loaded)
    {
        return;
    }

    std::string sourcePortName;
    if(m_parameters.check("name")){
      m_texName      = m_parameters.find("name").asString();
    } else {
      m_texName      = "/"+m_model->GetName();
    }
    if (yarp::os::NetworkBase::exists(m_texName))
    {
        m_texName = m_texName + "_2";
    }

    m_VideoPort.open(m_texName);
    m_VideoPort.setReadOnly();

    m_connection   = event::Events::ConnectPreRender(std::bind(&VideoTexture::Update, this));
    m_width        = 480;
    m_height       = 640;
    m_scale        = 1;
    m_width        = m_parameters.find("widthRes").asInt();
    m_height       = m_parameters.find("heightRes").asInt();
    m_scale        = m_parameters.find("heightLen").asDouble();
    m_material     = Ogre::MaterialManager::getSingleton().getByName(m_model->GetMaterialName());
    m_texture      = Ogre::TextureManager::getSingleton().getByName(m_texName);

    if(m_texture.isNull())
    {
        m_texture = Ogre::TextureManager::getSingleton().createManual
                  (
                      m_texName,
                      Ogre::ResourceGroupManager::DEFAULT_RESOURCE_GROUP_NAME,
                      Ogre::TEX_TYPE_2D,
                      m_width, m_height,
                      0,
                      FORMAT,
                      Ogre::TU_DYNAMIC_WRITE_ONLY_DISCARDABLE
                  );
    }

    if(m_parameters.check("defaultSourcePortName"))
    {
        sourcePortName = m_parameters.find("defaultSourcePortName").asString();

        if(!sourcePortName.empty() && yarp::os::NetworkBase::exists(sourcePortName))
        {
            yarp::os::NetworkBase::connect(sourcePortName, m_texName);
            Update();
        }
    }
    else
    {
        yInfo()<<"[charles] parameter sourcePortName NOT FOUND !!!";
        unsigned char* data = new unsigned char[m_width * m_height * 3];
        for(size_t i = 0; i < m_width * m_height * 3; i++)
        {
            data[i] = std::rand() % 255;
        }
        m_texture->getBuffer()->blitFromMemory(PixelBox(m_width, m_height, 1, FORMAT, (void*)data));
    }

    if(m_material.get())
    {
        m_material.get()->getTechnique(0)->getPass(0)->removeAllTextureUnitStates();
        m_material.get()->getTechnique(0)->getPass(0)->createTextureUnitState(m_texture->getName());
        m_material.get()->setReceiveShadows(false);
    }

    //double wScale = m_scale*(float(m_width)/float(m_height));
    //m_model->GetSceneNode()->scale(wScale, m_scale, m_scale);
    //m_model->GetSceneNode()->translate(0, m_scale/2, 0);

  }

  GZ_REGISTER_VISUAL_PLUGIN(VideoTexture);
}
```

pattacini, 2017-05-30:
Uhm @charleswilmot would you mind reposting your long snippet on Gist instead, and replacing the above comment with a pointer to it? This way we can keep the issue less cluttered. Thanks!

aerydna, 2017-06-14:
i'm on the issue.. when loading the default sdf no problem appears.. by opening your sdf ogre (and so gazebo) crashes. so probably, underneath, there are more serious issues than port conflicts. however i think fits better in the issue section of gazebo-yarp-plugin repository (https://github.com/robotology/gazebo-yarp-plugins) than Q&A

pattacini, 2017-06-14:
Issued moved as @aerydna suggested.

@aerydna @charleswilmot please comment on the new issue then.

Closing this.

closed 2020-12-09 by pattacini