robotology/community #223: Force control on iCub Head in Gazebo

Hteja, 2017-05-19:
Hi,

I am trying to perform motor babbling on the **iCub Head and Eyes** to generate data for learning the forward dynamic models of head-eye system in Gazebo. For this purpose, I need to apply torque commands on the joints directly and record the state of the head (position , velocity). 
1. However, I do not want the movements of rest of the body to effect the head/eye system when I apply high torques to the head/eye system. Could you please let me know how can this be achieved?
2. Secondly, is there a way I can simulate the conditions of the experiments on real robot (i.e., **gravity must be ON**, only the head must be suspended freely in the simulated world without any gravity or position compensators, while the body mains rigid without collapsing due to gravity)?

Thank you very much for being helpful through this forum.

Best,
Hari Teja.

francesco-romano, 2017-05-19:
Hi @Hteja 

1. If you control in torque only the head, the rest of the body remains in position mode. 
The position PIDs should be able to keep the robot still.
2. head and hands do not have torque support on the real robot. I am afraid you have to choose a different set of joints

Hteja, 2017-05-19:
Hi @francesco-romano 
Thank you very much. I have tried your suggestion on the simulation, and it would be sufficient for my requirements (for the rest of the body to remain in position mode). 
I will use the iCub arm instead of head for torque control.

francesco-romano, 2017-05-19:
@Hteja ok. The first 4 joints of the arm (with a possibility also for the 5th) are torque controlled also on the real robot.

closed 2020-12-08 by pattacini