robotology/community #224: How to change the soft Min and Max position Limits of the hip roll joints

MichaelBombile, 2017-05-22:
Hi @DanielePucci, We are working on iCubLausanne02, where we are testing a walking algorithm. However, we noticed that the hip roll joints positions are limited within a range of [0-90] (even in the yarpmotorgui) and all commands value less than 0 are truncated, while their kinematics say that their range is from [-119  +17].  How could we change those limits (essentially the Min limits)?

Thanks in advance.

DanielePucci, 2017-05-22:
@MichaelBombile, more generally, most of configuration files associated with your robot is available at 

https://github.com/robotology/robots-configuration/tree/master/iCubLausanne02

 
To modify joint limits, and specifically the hip joints, you have to change the first two columns of the tags `jntPosMax` and `jntPosMin` in 

https://github.com/robotology/robots-configuration/blob/master/iCubLausanne02/hardware/motorControl/right_upper_leg-ems8-mc.xml#L15

and

https://github.com/robotology/robots-configuration/blob/master/iCubLausanne02/hardware/motorControl/left_upper_leg-ems6-mc.xml#L14

The first two columns associated with the above tags correspond to the `roll` and `pitch` hip joints
(see e.g. http://wiki.icub.org/wiki/ICub_joints#Left_leg). I strongly suggest you not to modify source files, but to create  a `.local` directory after importing the configuration files by means of `yarp-config` command, e.g. by typing 

~~~
$ yarp-config robot --import $YARP_ROBOT_NAME
~~~

MichaelBombile, 2017-05-22:
Thanks very much @DanielePucci, I will try it and I will let you know.

MichaelBombile, 2017-05-22:
Hi  @DanielePucci, I configured the files as you suggested and it worked. 
Thanks again.

closed 2020-12-08 by pattacini