robotology/community #226: iGazeControl: difference between setStabilizationMode and setTrackingMode

valeriosperati, 2017-05-30:
Hi,
I'm running the tutorial of iGazeController on the iCubSim (tutorial_gaze_interface).
Here it is used the function "setTrackingMode", which should compensate gaze when torso is moved.
In the [webpage description](http://wiki.icub.org/brain/icub_gaze_interface.html) it is described the function "setStabilizationMode", which as I understand, should do the same work. It is not clear to me:
1) the difference between them IF the robot DOES NOT WALK
2) if one function excludes the other

Thanks, Valerio

pattacini, 2017-06-01:
Hi @valeriosperati 

The two modalities elicit two different pathways as described below.

#### Tracking
When the robot has attained its target, the tracking mode simply **triggers the controller again** to keep the focus on the desired location as soon as the distance between the current fixation point and the target overcomes a given threshold.

Two things to bear in mind: 
- The target position flows through the entire controller's chain, including the Cartesian solver.
- The retrigger may happen because the target has moved or e.g. the torso is being used for other tasks.

#### Stabilization
To stabilize the gaze we can use the **gyro readouts**, which differently from the case above, do not feed the whole controller's chain but rather act more downstream (on the **velocities**) as shown in the sketch below (more details can be found in [1] and [2]).

[image]

Therefore, the **reaction response is much faster** than the case of the compensation merely based on the position error seen by the controller between the target and the current fixation point.

Of course, **you can use both modalities at the same time**. To appreciate the benefit you can gain enabling stabilization, watch this video:

[image]

During stabilization, the head does not lag behind the target thanks to the compensation produced by the gyro feedback.

---
#### References
[1] "[_Gaze stabilization for humanoid robots: A comprehensive framework_](https://doi.org/10.1109/HUMANOIDS.2014.7041369)", A. Roncone, U. Pattacini, G. Metta, L. Natale, **International Conference on Humanoid Robots**, Madrid, Spain, 2014.

[2] "[_A Cartesian 6-DoF Gaze Controller for Humanoid Robots_](https://doi.org/10.15607/RSS.2016.XII.022)", A. Roncone, U. Pattacini, G. Metta, L. Natale, **Proceedings of Robotics: Science and Systems**, Ann Arbor MI, 2016.

valeriosperati, 2017-06-01:
Thanks a lot!

closed 2020-12-08 by pattacini