robotology/community #227: [gazebo + iCub] Simulating iCub faster than realtime

emilio-cartoni, 2017-06-05:
I would like to simulate the iCub on Gazebo faster than realtime, but it seems the plugins cannot work faster than 1kHz (realtime).

I have followed the tips here (#161) to achieve a fast simulation on Gazebo: namely I have removed all iCub parts except the torso and the left arm.
Gazebo now operates in realtime (Real Time Factor 1.00) instead of running at 0.30 when simulating the whole iCub.
However, I need to run very long simulations and a real time factor (RTF) higher than 1 would be highly desiderable.
Given that it's not possible to increase the max_step_size in Gazebo without messing with the PIDs (see #161), I've tried to increase the real_time_update_rate in Gazebo from 1.000 to 10.000.
The result is that Gazebo now runs at 3.5 Real Time Factor (limited by the CPU, in theory with 10k update rate and 0.001 step size it could go up to 10 RTF).
However, when running faster than real time, the iCub stops responding.
When the real_time_update rate is higer than 1000, if I connect to the iCub with yarpmotorgui or using some other controller (i.e. a custom c++ program) everything looks fine but the iCub does not move.
If I restore the real_time_update_rate to 1000, I can again move the iCub from the motorgui (without restarting the gui or anything else).
Even if I set a real_time_update_rate of 1001 (thus staying at basically 1.00 RTF) the iCub cannot be moved.
Interestingly, if I set  a real_time_update_rate of 500, thus slowing down the simulation to 0.5 RTF, I can move the iCub.
So it seems that the controllers are designed to work at a maximum of 1000 Hz - even 1001 Hz won't work.

Is there any workaround or setting to be able to achieve faster than real-time iCub simulations?

traversaro, 2017-06-05:
Hi @emilio-cartoni , 
  what you are describing seems indeed a bug, either in `gazebo`, in `gazebo-yarp-plugins` or in something YARP related. I will then move this issue to `gazebo-yarp-plugins` repository, that is probably the piece of software most related to this problem.

traversaro, 2017-06-05:
Issue moved to [robotology/gazebo-yarp-plugins #300](https://github.com/robotology/gazebo-yarp-plugins/issues/300) via [**ZenHub**](https://www.zenhub.com/)

closed 2020-12-09 by pattacini