robotology/community #228: [ANNOUNCEMENT] 2017-06-15 - Release Day

drdanz, 2017-06-14:
Tomorrow, 2017-06-15, we are going to release several robotology "core" packages (RTF, YCM, YARP,  and iCub-*).
We are going to merge `devel` into `master` for all the repositories, therefore there could be issues if you are on the `master` branch and you update the repositories while the some are merged and other are not.
Therefore we suggest not to pull the master branch until the releases are done (announcements will follow).

@robotology/all-robotology-developers

traversaro, 2017-06-16:
For anyone reading: the release process is still ongoing at the moment. 
 cc @gabrielenava @S-Dafarra

francesco-romano, 2017-06-16:
I would reopen this issue until we finish the process @pattacini

drdanz, 2017-06-16:
Everything was released.
Closing this issue

closed 2020-12-12 by pattacini