robotology/community #229: gcc version

amaroyo, 2017-06-15:
Hi, 

I am using macOS and I was using the function ```std::to_string(myDoubleVar);``` which is working fine. However, I was compiling it in the linux setup and it's not recognized **'to_string' is not a member of 'std'** 

It seems that the function is C++11, but I thought we were using already the [C++11](https://github.com/robotology/QA/issues/173). 

Any ideas? Do we need to upgrade the linux setup with a newer version? 

Thanks, 
Alex.

pattacini, 2017-06-15:
Hi @amaroyo,

C++11 is not enabled by default on Linux, or at least on certain flavors of Linux; therefore, you have to explicitly tell CMake to compile the code with C++11 features enabled.

There are [multiple ways](https://stackoverflow.com/questions/10851247/how-to-activate-c-11-in-cmake) of doing this. If you have CMake 3.1, try this:

```cmake
set (CMAKE_CXX_STANDARD 11)
```

traversaro, 2017-06-15:
C++11 (actually C++14) will be the default from GCC 6.1 (https://gcc.gnu.org/projects/cxx-status.html#cxx14), that I think is the default since Ubuntu 16.10 .

closed 2020-12-15 by pattacini