robotology/community #240: How to update you robot to use iCub 1.8.0

marcoaccame, 2017-06-23:
Two posts of last year give instructions about how to update an ETH robot to the latest version of iCub software: https://github.com/robotology/QA/issues/134  and https://github.com/robotology/QA/issues/135.

Since then, some things have changed: we have the xml files inside repository  https://github.com/robotology/robots-configuration and we have a new tool called `FirmwareUpdater` to perform FW update (see `NOTE 1`).

As a result of that, here I post an updated procedure which applies to both CAN and ETH robots.

On server:
1. Pull latest master version of yarp and compile it. 
2. Pull latest master version of icub-main and compile it.

On pc104 or icub-head (see `NOTE 2`):
1. Pull latest master version of yarp, icub-main, icub-firmware-shared, icub-firmware-build and robots-configuration. 
2. Compile yarp.
3. Compile icub-firmware-shared: `cd icub-firmware-shared/build-pc104`, `ccmake ..`, `make`.
4. Compile icub-main: `cd icub-main/build-pc104`, `ccmake ..`, `make`.
5. Install the robot-configuration files. See instructions in https://github.com/robotology/robots-configuration.
6. Perform FW update of the ETH boards (EMS/MC4PLUS/MC2PLUS) and of the CAN boards with the latest available FW releases.  See instructions in https://github.com/robotology/icub-firmware-build/blob/master/FirmwareUpdater.readme.quick.txt

If one follows this procedure the robot will be ready to execute correctly.

In case, however, one misses some points or if forgets to perform FW update of some boards, proper messages from cmake and diagnostics printed when running yarprobotinterface will guide to perform missing update actions.

---
`NOTES`

1. For info about the new tool called `FirmwareUpdater`  which was released in icub-main master only a few days ago, please read: 
   - https://github.com/robotology/QA/issues/221, 
   - https://github.com/robotology/icub-firmware-build/blob/master/FirmwareUpdater.readme.fulldetails.txt, 
   - https://github.com/robotology/icub-firmware-build/blob/master/FirmwareUpdater.readme.quick.txt,
   - https://github.com/robotology/icub-firmware-build/blob/master/ETH/FirmwareUpdater.readme.maintenance.of.eth.robots.txt)
2. Some new robots have a hard disk internal to the PC104 and do not mount the disk in NFS anymore. That is the reason why instructions specify to perform the pull on server but also on PC104.
3. For guide about how to configure the YARP and ICUB compilation, please refer to http://wiki.icub.org/wiki/Linux:Installation_from_sources.

plinioMoreno, 2017-06-23:
Here in Lisbon we would be very interested in setting this tool for our robot Vizzy. From what I read, it looks like we have to define a ```firmwareupdater.ini``` file. Is that just the only file we need, @randaz81 and @marcoaccame?

gsaponaro, 2017-06-23:
Hi @marcoaccame, thanks for the clear instructions.

One comment related to step 5 (installing the robots-configuration files):

the motor control xml files of the iCubs outside of Genoa are lacking the new mandatory motor control parameters `fullscalePWM` and `ampsToSensor`. Unfortunately this means that, after following the update instructions, `yarprobotinterface` will not start. I opened an issue specific to iCubLisboa01 [here](https://github.com/robotology/icub-support/issues/483), but perhaps it is worth fixing those xml's for all non-Genova robots? I found out [this commit](https://github.com/robotology/robots-configuration/commit/0e9e77ec74191782b5bd3bba461fcdef17924893), where the comment said to propagate those parameters for all robots, but AFAIK this was not done.

traversaro, 2017-06-28:
Please note that, following #173, if you want to use the iCub software version 1.8.0 on a physical iCub mounting a pc104 board, you will need to update the OS running on the pc104 to the 6.0 image based on Debian Jessie. Please refer to http://wiki.icub.org/wiki/The_Linux_on_the_pc104 for more information .

dogoepp, 2017-06-28:
We are following the procedure to update iCubNancy01. Right now we are updating the firmwares and I wonder if the eUpdater and eLoader should be upgraded too.

@gabrielenava @pi-q

marcoaccame, 2017-06-29:
Hi @dogoepp, I would recommend performing a full update of your robot with also the latest versions of eUpdater and eLoader. But do that with care. Read https://github.com/robotology/QA/issues/221 for more info.

marcoaccame, 2017-06-29:
Hi @gsaponaro, sorry for my late reply but I had a few days holiday. 
About xml files for iCubLisboa01: I was aware of your post and last week I alerted the relevant persons to help you solve your problem. 

Usually we add new parameters in xml files of all the robots. However, for the case of `fullscalePWM` etc the  propagation to all robots got lost. sic.

marcoaccame, 2017-06-29:
Hi @traversaro, thank you for your contribution. I have updated my original post with your comment to update the OS image on PC104.

marcoaccame, 2017-06-29:
Hi @plinioMoreno, I am sorry I missed to add a firmwareupdater.ini file in the vizzy folder.

I will do it for you soon and let you know,

To compile the FirmwareUpdater tool you need to update the system to iCub 1.8.0.

marcoaccame, 2017-06-29:
Hi @plinioMoreno, I have updated the vizzy robot with firmwareupdater.ini.

closed 2020-12-08 by pattacini