robotology/community #241: delete virtual objects in Gazebo through worldInterface plugin

valeriosperati, 2017-07-04:
Hi, 
I successfully created a cylinder in gazebo using the worldInterface plugin.  
I did it connecting my c++ code to  the port "/world_input_port", and then writing those commands, as suggested in [#201](https://github.com/robotology/QA/issues/201) by @pattacini :

```
yarp::os::Bottle cmd, rep;

cmd.addString("makeCylinder");
cmd.addDouble(0.05); //radius [mt]
cmd.addDouble(0.1);  //lenght x  [mt]
cmd.addDouble(1.4);  //pos x  [mt]
cmd.addDouble(0.8);  //pos y	    
cmd.addDouble(0.3);  //pos z
cmd.addDouble(0.1);  //pose.roll  [rad] 
cmd.addDouble(0.1);  //pose.pitch [rad]	    
cmd.addDouble(0.1);  //pose.yaw   [rad]
cmd.addInt(0);     //color.r 
cmd.addInt(255);   //color.g 
cmd.addInt(0);     //color.b 

portToGazeboWorld.write(cmd, rep);
```

Now, the problem is that I cannot delete the object. Looking at [**Thrift**](https://github.com/robotology/gazebo-yarp-plugins/blob/master/thrift/worldinterface/worldinterface_rpc.thrift)  interface I wrote this command at the end of my program:

```
yarp::os::Bottle cmd, rep;
cmd.addString("deleteAll");           
portToGazeboWorld.write(cmd, rep);
std::string response=rep.get(0).asString();  
cout<<"Operation: "<<response<<"\n\n";
```
I guess the syntax is correctly received by gazebo, as the response is 
`Operation: ok`

Nevertheless, the cylinder remains in gazebo. What am I missing?

PS: I noted in the gazebo terminal a message appearing as soon as the object is created
`[WARNING]Internal error during object creation. Unimplemented feature in gazebo 7.`

valeriosperati, 2017-07-04:
I realised right now it happens with cylinder object only. Speher and boxe behave as expected.
I get non name back from cylinder, while I get names from sphere and box.

pattacini, 2017-09-25:
Hi @valeriosperati 

Apologies for the very long delay in getting back to you. The thrift API deserves a careful inspection in this respect, hence I'll move this issue to the corresponding repository.

pattacini, 2017-09-25:
Issue moved to [robotology/gazebo-yarp-plugins #320](https://github.com/robotology/gazebo-yarp-plugins/issues/320) via [**ZenHub**](https://www.zenhub.com/)

closed 2020-12-09 by pattacini