robotology/community #242: simulated icub on gazebo drifts when using Gaze controller

valeriosperati, 2017-07-10:
Hi, I have a code that initialises the gaze controller on simulated icub and, on a keypress event, directs the gaze on a fixed point using lookAtMonoPixel() function.
It works as expected with icub_SIM but fails using gazebo: the head robot immediately drifts after the initialisation of gaze controller

[video example](https://drive.google.com/open?id=0B7eFeb2P2Ie7bUV6NVFGNUlrb00)

 ```
    PolyDriver m_clientGaze;
    Property option_gaze;
    option_gaze.put("device","gazecontrollerclient");
    option_gaze.put("remote","/iKinGazeCtrl");
    option_gaze.put("local","/client/gaze");
    
    if(!m_clientGaze.open(option_gaze))
    {
	cout<<"\n[FATAL], problems in opening gaze controller...\n\t";
	cout<<"Is iKinGazeController running?...\n\n";
        std::string exit(0);
    }
    
    if(m_clientGaze.isValid())
    {
	m_clientGaze.view(iGaze);
    }else
    {
	cout<<"\n[FATAL], problems in viewing gaze controller...\n";
        std::string exit(0);
    }
    
    iGaze->setNeckTrajTime(0.8);
    iGaze->setEyesTrajTime(0.4);
    
    // put the gaze in tracking mode, so that
    // when the torso moves, the gaze controller 
    // will compensate for it
    iGaze->setTrackingMode(true);
    iGaze->setStabilizationMode(true);
```

iKinGazeCtrl is launched with parameters:

`iKinGazeCtrl --robot icubSim --from configSim.ini`

Gazebo is launched  with parameters:

`gazebo setupV0.sdf`

where setupV0.sdf is my starting configuration running  icub_with_hands and the worldInterface plugin:

```
<?xml version='1.0'?>
<sdf version='1.4'>
  <world name="my_world">

    <!-- Light -->
    <include>
      <uri>model://sun</uri>
    </include>

    <!-- Ground Plane -->
    <include>
      <uri>model://ground_plane</uri>
    </include>

   <!-- icub  and worldinterface plugin -->
   <model name="iCub_with_hands">
   	<include>
	    <uri>model://icub_with_hands</uri>
    	</include>
        <plugin name="world_interface" filename="libgazebo_yarp_worldinterface.so">
           <yarpConfigurationFile>model://worldInterface/conf/worldInterface.ini</yarpConfigurationFile>
[... 535 lines cut ...]
yarp: Receiving input from /iKinGazeCtrl/head/rpc:o to /icubSim/head/rpc:i using tcp
yarp: Receiving input from /iKinGazeCtrl/head/command:o to /icubSim/head/command:i using udp
yarp: Sending output from /icubSim/head/stateExt:o to /iKinGazeCtrl/head/stateExt:i using udp
[INFO]Read [41148] frames in 5[s], average period 0.12[ms], min 0.06[ms], max 3.12[ms]
[INFO]Read [40868] frames in 5[s], average period 0.12[ms], min 0.06[ms], max 2.49[ms]
[INFO]Read [40378] frames in 5[s], average period 0.12[ms], min 0.06[ms], max 4.22[ms]
[INFO]Read [40825] frames in 5[s], average period 0.12[ms], min 0.06[ms], max 4.11[ms]
yarp: Sending output from /icubSim/cam/right to /m_Port_rightCam using tcp
yarp: Receiving input from /head_client/rpc:o to /icubSim/head/rpc:i using tcp
yarp: Receiving input from /head_client/command:o to /icubSim/head/command:i using udp
yarp: Sending output from /icubSim/head/stateExt:o to /head_client/stateExt:i using udp
[INFO]Read [34576] frames in 5[s], average period 0.14[ms], min 0.06[ms], max 3.85[ms]
[INFO]Read [29283] frames in 5[s], average period 0.17[ms], min 0.07[ms], max 5.37[ms]
yarp: Removing input from /head_client/rpc:o to /icubSim/head/rpc:i
yarp: Removing input from /head_client/command:o to /icubSim/head/command:i
yarp: Removing output from /icubSim/head/stateExt:o to /head_client/stateExt:i
yarp: Removing output from /icubSim/cam/right to /m_Port_rightCam
[INFO]Read [35578] frames in 5[s], average period 0.14[ms], min 0.06[ms], max 2.69[ms]
[INFO]Read [31172] frames in 5[s], average period 0.16[ms], min 0.06[ms], max 12.31[ms]
[INFO]Read [41033] frames in 5[s], average period 0.12[ms], min 0.06[ms], max 1.97[ms]
[INFO]Read [41400] frames in 5[s], average period 0.12[ms], min 0.06[ms], max 1.08[ms]
```

Thanks for help,
Valerio

pattacini, 2017-07-10:
@valeriosperati I'm afraid that what you've seen is very much related to a number of open issues on https://github.com/robotology/icub-gazebo/issues that stem from the fact that `icub-gazebo` still need quite a lot of tuning of the low-level PID control layer.

See for example https://github.com/robotology/icub-gazebo/issues/38

Therefore, if you don't mind, I'd rather move this issue therein.

valeriosperati, 2017-07-11:
Yea, sure!

pattacini, 2017-07-11:
Issue moved to [robotology/icub-gazebo #61](https://github.com/robotology/icub-gazebo/issues/61) via [**ZenHub**](https://www.zenhub.com/)

closed 2020-12-09 by pattacini