robotology/community #243: Gaze interface: retrieving 3D fixation point using monocular approach

valeriosperati, 2017-07-12:
Please, I need some clarification about the issue object.
In icub simulator (icub_SIM), I'm successfully using lookAtMonoPixel function in order to gaze a ball on a table with right cam . The ball has coordinates x=0, y= 0.7; z=1.5 in world reference frame (these should be about x=-1.5, y= 0, z= 0.10 in robot reference frame).
Once the gaze is performed, the object is correctly located at the centre of image, about 160, 120 pixels.

Now I would like to retrieve the 3D point of the object.
As I understand [here](http://wiki.icub.org/brain/icub_gaze_interface.html), "geometry of pixels" paragraph, I need to use the function get3DPointOnPlane(). Maybe I misunderstood the meaning of function arguments:  my code is 
```
Vector 3Dpoint;
Vector px(2);
px[0] = 160;
px[1] = 120;

Vector plane(4);
plane[0]=0;
plane[1]=0;
plane[2]=1;
plane[3]= 0.1;  // I guess this is the z of ball in robot reference frame

//using right cam
igaze->get3DPointOnPlane(1, px, plane, 3DPoint);

```
I would expect 3DPoint is about  [-1.5, 0, 0.1], but I obtain something like [-3, 0, -0.1]. What Am I missing?

vicentepedro, 2017-07-12:
Hi @valeriosperati,

You can check [here](http://wiki.icub.org/brain/icub_gaze_interface.html#subsec_gaze_geometrypixels)
in the first code box.

in your code it should be:
`plane[3]= -0.1;`
since you are solving the plane equation:
`ax+by+cz+d=0`

the equation will become:
`z-0.1=0 <=> z=0.1`

valeriosperati, 2017-07-13:
Ops! I completely missed the minus 1! Thanks a lot :) it works

closed 2020-12-09 by pattacini