robotology/community #246: iCub SEA legs - Hard and Soft joint limits

fiorisi, 2017-08-17:
Which are the hardware and software joint limits for the iCub SEA legs?

fiorisi, 2017-08-17:
The issue collects the __HARD__ and the __SOFT__ joint limits for the iCub `SEA legs` (Series Elastic Actuator legs).
The SEA legs are generally assembled on all version of *iCub 2.5* and on the first version of the *Koroibot* robot.  
The joint limits are defined as follows:
* The HARD joint limits are enforced at the *joint level* through a mechanical hard stop (e.g. a pin).
The joint limits have been extracted from the CAD model of the left leg.
The assembly name is: `RC_IIT_018_A_012`.
* The SOFT joint limits are enforced at the *firmware level*.
The values have been copied from the [`iCubGenova04 configuration file`](https://github.com/robotology/robots-configuration/tree/devel/iCubGenova04/hardware/mechanicals).

The following table recaps the joint limits for the complete leg:

|               |    HARD     |     HARD    | SOFT | SOFT |
|---------------|-------------|-------------|-------------|-------------|
| JOINT NAME    | UPPER LIMIT [deg] | LOWER LIMIT [deg] | UPPER LIMIT [deg] | LOWER LIMIT [deg] |
| hip_pitch     |     134     |     -45     | 85 | -30 |
| hip_roll      |     120     |     -20     | 85 | -15 |
| hip_yaw       |      80     |     -80     | 70 | -70 |
| knee          |      4      |     -124    |  0 |-100 |
| ankle_pitch   |      35     |     -35     | -30 | 30 |
| ankle_roll    |      25     |     -25     | 20 | -20 |

## HARD joint limits
### l_hip_pitch
[image]
[image]

### l_hip_roll
[image]
[image]

### l_hip_yaw
[image]

### l_knee
[image]
[image]

### l_ankle_pitch
[image]
[image]

### l_ankle_roll
[image]

fiorisi, 2017-08-17:
cc: @traversaro @DanielePucci @francesco-romano @gabrielenava @S-Dafarra @pattacini 
related issues: 
* https://github.com/loc2/element_icub-at-ihmc/issues/2
* https://github.com/robotology-playground/icub-model-generator/issues/48
* https://github.com/robotology-playground/icub-model-generator/issues/5

DanielePucci, 2017-08-17:
Hi @fiorisi, thanks a lot for the effort. Crystal.

Did you by chance check that these limits are met also on a "real" robot we have in the lab?

fiorisi, 2017-08-17:
The joint limits have been checked on the real robot. The problem is that in this case it is necessary to take into account cover self-collisions.
The joint limits have been measured with the Yarp robot motor GUI from the left leg of the `iCubGenova04` robot.

The following table recaps the joint limits for the complete leg:

|               | REAL ROBOT | REAL ROBOT |
|---------------|-------------|-------------|
| JOINT NAME    | UPPER LIMIT [deg] | LOWER LIMIT [deg] |
| hip_pitch     |     100     |     -20 \*     |
| hip_roll      |      85     |      -10 \*    |
| hip_yaw       |     55 \*    |       -55 \*   |
| knee          |      4     |     -110    |
| ankle_pitch   |     34     |      -34 \**    |
| ankle_roll    |     25     |      -25    |

__(\*) NOTE__: the joint limits of the hip have complex interdependencies. If the SOFT limits are used, when the joint *hip_yaw* is rotated close to its limit (either 70 or -70), the covers can collide with the waist when moving the other hip joints (namely, *hip_roll* and *hip_pitch*). For this reason the REAL ROBOT limits have a smaller range of motion.

__(\**) NOTE__: in the CAD model, the joint has a cover collision before reaching the limit value. Nevertheless, probably due to deformations during the production of the covers, there is no collision on the iCubGenova04.  
This problem will be fixed ASAP in the CAD model.

### l_hip_pitch
[image]
[image]

### l_hip_roll
[image]
[image]

### l_hip_yaw
[image]
[image]

### l_knee
[image]
[image]

### l_ankle_pitch
[image]
[image]

### l_ankle_roll
[image]
[image]

DanielePucci, 2017-08-18:
Nice. I think we can also update the corresponding urdf models then

@traversaro @francesco-romano @pattacini

traversaro, 2017-08-18:
The file to update is https://github.com/robotology-playground/icub-model-generator/blob/master/simmechanics/data/icub2_5/ICUB_2-5_BB_joint_parameters.csv . 

We should also update the iCubHeidelberg01 model, but that one is still not integrated in the repo for some issues: 
* https://github.com/robotology-playground/icub-model-generator/pull/15
* https://github.com/robotology-playground/icub-model-generator/issues/35 
* https://github.com/robotology-playground/icub-model-generator/issues/16

serena-ivaldi, 2017-11-04:
That applies also to our iCubNancy01, right? In that case we should update also our URDF, right? Can we assume the same values for the covers (collisions)?

@dogoepp @InriaBrice @olivierrochel-inria

pattacini, 2017-11-06:
@fiorisi could you take on @serena-ivaldi's question?

randaz81, 2017-11-06:
Please open a new issue for your robot

S-Dafarra, 2018-04-12:
After a double check on the robot with @fiorisi we noticed that the limits were bigger than what written initially. The table in the second comment has been updated together with the corresponding CAD images. This time the range has been measured by constraining the pin to be coincident with the hard limit surface (initially the range of motion was measured considering only the hollow, but the pin as a smaller radius, causing the inconsistency noticed above). 

Notice that it is also possible to reach all the leg joints hard limits, except from the knee lower limit due to the cabling. All the other limits are more or less trivially reachable even with the covers mounted. The tricky ones to be reached are the hip roll limits. The upper limit can be reached by bending the leg in this pretty awkward position, where also the pitch and the yaw are close to their limits:
[image]

The hip roll lower limit instead can be approached when the robot is **not** on the pole, even if it cannot be mechanically reached because the upper leg is hitting the hip_pitch pulley:
[image]
[image]

closed 2020-12-15 by pattacini