robotology/community #247: Saccadic eye movements - icubSim

NunoDuarte, 2017-08-21:
Hi,

I'm trying to develop an experiment where the icub needs to perform some actions involving grasping and placing the object in a different location. During an action I would like the icub to first look at the end location with its eyes and then move its head and body, just like a human would. 

I noticed that there is a very interesting article where you tackle this. 
http://www.roboticsproceedings.org/rss12/p22.pdf 
Implement on the ikingaze controller.
There is also a video showing it working on the real robot: https://www.youtube.com/watch?v=I4ZKfAvs1y0&feature=youtu.be 

For some reason every time I run in the simulation the icub doesn't seem to change the eye gaze very much and what I get is head gaze instead. 

My question is if what I read in the article (and I saw in the video) is possible to obtain the same behavior in the icub Simulator or if it is not implemented yet.

I tried to turn saccadic movements on but with no success.
`igaze->setSaccadesMode(true); `

Thanks in advance,

Nuno Duarte 

/cc maybe @pattacini knows the problem I am facing.

pattacini, 2017-08-23:
Hi @NunoDuarte 

Unfortunately, the saccadic behavior is not implemented in the iCub simulator 😞 

What you could do, instead, in order to make the initial eyes movements stand out compared with the head movements is simply to reduce the reactivity of the neck joints by means of the API method [`IGazeControl::setNeckTrajTime()`][1]. To this purpose, just pass an increased neck trajectory time to this function (i.e. something like `> 1.0` seconds).

[1]: http://www.yarp.it/classyarp_1_1dev_1_1IGazeControl.html#a14b3e2c61d2b1568ea05dbf9ebd9e4b3

NunoDuarte, 2017-08-23:
Hi @pattacini 

Thank you for the quick response.

Yes, I thought so too. 
I already tried setting the neck trajectory time be bigger than the eyes trajectory time but I didn't get a very big difference. However, when I tried on the real robot I got the result I was expecting, even with the saccades turned off. 

Thanks for the information,
Nuno

PS: How about the Gazebo simulator? Does it have the saccadic behavior implemented?

pattacini, 2017-08-23:
Did you try to significantly increase the neck trajectory time? Anyway, I don't think it's, in the end, a big problem, given that it works on the real robot.

What is required to implement the saccades is the so-called **MIXED** mode (see #169). I don't know if it is available in Gazebo already, although I believe we should open a different thread for this.

NunoDuarte, 2017-08-24:
For example, I tried:

igaze->setEyesTrajTime(0.05);
igaze->setNeckTrajTime(3.0);

And I didn't see saccadic movements but rather some small oscillations of the eyes combined with very slow head motion. It's as if the gaze controller prefers to move the neck rather than the eyes. I can keep trying different combinations to see if it shows any difference. 

As for Gazebo, I think for now it is not necessary to send an issue. 
Perhaps when we all (in VisLab) switch to icub-gazebo.

Thanks @pattacini. I got the answer I was looking for.

pattacini, 2017-08-24:
Setting the eyes trajectory time to `0.05` seconds is way too fast and dangerous 😉; that's the reason why you're observing those weird results.

What I did for a quick test was:
```sh
yarp rpc /iKinGazeCtrl/rpc
>> set Tneck 2.0
>> look 3D (-0.5 0.0 0.36)
>> look 3D (-0.5 -0.2 0.36)
>> look 3D (-0.5 0.0 0.36)
>> look 3D (-0.5 0.2 0.36)
```
thus leaving Teyes to its default value of `0.25` seconds and obtaining the following coordinated movements:

[image]

It seems to me that these movements are quite reasonable.

NunoDuarte, 2017-08-24:
Yeah, I just tried that! It is much better. 

I guess I should have only changed the neck trajectory time. 

Thanks again!

closed 2020-12-08 by pattacini