robotology/community #249: How to limit joint velocities in task velocity

NunoDuarte, 2017-09-12:
Hi everyone,

I'm using Cartesian interface to control in task velocities using the following function:

`icart->setTaskVelocities(xdot,odot)`

I was wondering if the joint velocities are limited or not and if I can set a limit so I won't break the iCub :)

Thanks in advance,
Nuno

pattacini, 2017-09-12:
Hi @NunoDuarte 

#208 does pretty much explain how `setTaskVelocities()` works.

To limit the reactivity of the controller, therefore, do play with the `Ttraj` parameters by calling [`setTrajTime()`][1]. Higher `Ttraj` correspond to slower joint velocities.

[1]: http://www.yarp.it/classyarp_1_1dev_1_1ICartesianControl.html#a46ede2ac6bed5e0e459deb843a8f7783

NunoDuarte, 2017-09-15:
Hi Pattacini,

Sorry for the late response. I saw your comment. I thought there was another way to control the joint velocities (like setRefVelocities) but I think that will do. :)

Thanks

pattacini, 2017-09-15:
No worries @NunoDuarte 

The idea is that you don't see the mechanism (i.e. joints) underneath, but rather you keep dealing with the end-effector only 😉 

How the joints move in terms of direction and speed is a detail that should remain hidden in the implementation.

closed 2020-12-08 by pattacini