robotology/community #254: Cartesian Interface in Simulator does not find yarprobotinterface.ini

pattacini, 2017-10-07:
@mpatacchiola commented on [Fri Oct 06 2017](https://github.com/robotology/icub-support/issues/546)

Dear all,

I am having trouble running the Cartesian interface on the simulator. The new procedure reported in the [official documentation](http://wiki.icub.org/brain/icub_cartesian_interface.html) says that to enable the controller on the simulator it is necessary to follow these steps:

1. Launch the iCub Simulator.
2. Deploy the Cartesian servers: yarprobotinterface –context simCartesianControl
3. Launch the Cartesian Solvers for the required limbs: have a look to the template located in the directory icub-main/app/simCartesianControl/scripts.

The problem arise in the second step, lunching that command I get the following error:

```
yarprobotinterface -context simCartesianControl
||| configuring
||| default config file specified as yarprobotinterface.ini
||| checking [/home/mpatacchiola/yarprobotinterface.ini] (pwd)
||| checking [/home/mpatacchiola/.config/yarp/yarprobotinterface.ini] (YARP_CONFIG_HOME)
||| checking [/home/mpatacchiola/.local/share/yarp/yarprobotinterface.ini] (YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/yarprobotinterface.ini] (YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/yarprobotinterface.ini] (YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/yarprobotinterface.ini] (YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/usr/share/yarp/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/var/lib/snapd/desktop/yarp/yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [/usr/local/share/iCub/yarprobotinterface.ini] (yarp.d)
||| did not find yarprobotinterface.ini
yarp: cannot read from yarprobotinterface.ini
[FATAL]Missing "config" argument 
yarprobotinterface() [0x41d690]
/usr/local/bin/../lib/libYARP_OS.so.1(_ZN4yarp2os8RFModule9runModuleERNS0_14ResourceFinderE+0x62) [0x7f56911f6902]
yarprobotinterface() [0x40583a]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7f569076c830]
yarprobotinterface(_start+0x29) [0x405be9]
```

It seems that the file `yarprobotinterface.ini` is missing. I guess that this file has a particular configuration that should enable the interface to work in simulation mode. I could not find the file in the repository and searching on Google did not get me any answer.

I tried to create an empty text file called `yarprobotinterface.ini` containing the line `config ./icub_all.xml` but this move the problem on the content of the `config ./icub_all.xml` file which seems to be tailored on a physical robot. Any suggestion is appreciated...

Kind regards, Massimiliano

---

@traversaro commented on [Fri Oct 06 2017](https://github.com/robotology/icub-support/issues/546#issuecomment-334867235)

I think the problem is that there is something wrong in the documentation, the correct command should be: 
~~~
yarprobotinterface --context simCartesianControl
~~~
with two "-" before context.

---

@pattacini commented on [Sat Oct 07 2017](https://github.com/robotology/icub-support/issues/546#issuecomment-334932382)

Hi @mpatacchiola 

As per our guidelines, this is not the correct issue tracking system for such a question. Thus, closing and moving to the right place.

pattacini, 2017-10-07:
@traversaro is right about the way you're required to specify command line parameters.
The documentation should be fine. I'm afraid doxygen renders `--` into a long dash.

mpatacchiola, 2017-10-07:
@pattacini Thank you for moving the post in the right section.

@traversaro Silly me, that was the problem! Using the command `yarprobotinterface --context simCartesianControl` everything works fine. Thanks for the suggestion ;-)

pattacini, 2017-10-07:
According to this [resource][1], there's no way to disable double dash interpretation in Doxygen via command line options. The only mean is to resort to escape character: `\--` 😞 

I've not found anything else as of now. 

[1]:  http://doxygen.10944.n7.nabble.com/Double-dash-within-lt-tt-gt-or-lt-code-gt-is-again-or-still-broken-td6590.html

closed 2020-12-15 by pattacini