robotology/community #255: yarpmanager "file changed"

arrenglover, 2017-10-11:
In the recent yarp devel branch yarpmanager detects if an application file (.xml) has been changed/updated and asks if you want it updated also in yarpmanager. I can understand in many situations this would be very helpful.

In my case I have a project that installs the application files (.xml) into the iCubContrib install directory such that they are detected by YARP manager automatically. Every time I compile my project with yarpmanager open (which is usually my workflow), yarpmanager opens the dialog box for "file changed" asking if I want to reload the application. In my project there are multiple applications (> 3), such that I every single time I recompile I have to select yes/no on the yarpmanager dialog box for each application. It is becoming frustrating.

I was wondering if anyone had the same problem and has a good solution? Maybe I there is a nice way to provide an option to "make install <option>" to only install .bin files? (ideally without having to change cmake options). Should I be only using TEMPLATE .xml files rather than directly installing .xml into iCubContrib? Is the best solution to have the option to turn off YARP module manager's automatic reloading of applications?

Thanks for any help.

pattacini, 2017-10-11:
Hi @arrenglover 

The normal workflow is to let `yarpmanager` pick up the xml files from `~/.local/share/yarp/applications`.

Installed templates, as they come, are not immediately suitable to be picked up by `yarpmanager` (they need to be customized), which in turn shouldn't be looking into installed directories.

arrenglover, 2017-10-11:
thanks @pattacini,
So in my cmake i use the `yarp_install()` macro to install the .xml files into `${iCubContrib}/share/ICUBcontrib/applications`. 

As I understand, you recommend that I should remove this automatic install of the application and instead manually copy the application files to `~/local/share/yarp/applications`?

pattacini, 2017-10-11:
Well, `yarpmanager` can certainly look into `${ICUBcontrib}/share/ICUBcontrib/applications` but shouldn't be instructed to seek into `${iCubContrib}/share/ICUBcontrib/template/applications`, which is instead the location where you should `yarp_install` your templates (not finalized applications).

arrenglover, 2017-10-11:
I agree - I am indeed not asking `yarpmanager` to looking into `${iCubContrib}/share/ICUBcontrib/template/applications` for applications. 

I am installing applications into `${iCubContrib}/share/ICUBcontrib/applications` and `yarpmanager` finds them nicely. My problem is that every time I build and install my project it installs the applications again and `yarpmanager` now notices the file has been changed and asks if I want to reload them. This is the problem because I have to press 'yes/no' for each application (>3) every time I recompile the code.

Should I not be using the `yarp_install()` macro to automatically install the applications into `${ICUBcontrib}/share/ICUBcontrib/applications`? I really find it nice to do this though.

pattacini, 2017-10-11:
Keep using `yarp_install` but rather than installing applications, you should be installing templates in the correct place 😄

arrenglover, 2017-10-11:
okay so:
1. `yarp_install()` template applications into the `${ICUBCONTRIB_APPLICATIONS_TEMPLATES_INSTALL_DIR}`
2. manually copy the template into `~/.local/share/yarp/applications`, renaming the file to .xml and filling in required information (e.g. which host to run the application)

correct?

pattacini, 2017-10-11:
Exactly!

The second step could be automatized somehow, as far as I know, directly from within the `yarpmanager` GUI, which thus should be capable of reading a template, asking for customization and saving in right location.

arrenglover, 2017-10-11:
thanks very much @pattacini :+1: 
@vvasco and @Iaxama - I'll update the CMake and move .xml to .template ASAP, so beware of this change in workflow :smile:

drdanz, 2017-10-11:
CC @Nicogene

Nicogene, 2017-10-12:
Sorry @arrenglover if yesterday I could not answer to this issue, but @pattacini explained you very well how to deal with `yarpmanager` xmls. 
If you put the templates in the folder `*template/applications` it should be ok, I think it is the best way to proceed, I don't like very much the idea of  the button to temporary disable the automatic reload.
If you have further questions or problems don't hesitate to contact me direclty :wink:

arrenglover, 2017-10-12:
in that case @Nicogene, the GUI-based method to convert an installed template into a local application seems like a really nice way to make applications accessible.

Nicogene, 2017-11-02:
Hi@apaikan,
the feature @pattacini is talking about is already implemented in `yarpmanager` ?

apaikan, 2017-11-02:
If you right click on a template file name from gui, you wont see the option to install it?

Nicogene, 2017-11-02:
[image]
Do you mean right click on these names?

apaikan, 2017-11-02:
yes!  nothing?

Nicogene, 2017-11-02:
Nope :pensive:
I have only the `Edit` option that opens gedit(on Ubuntu)

pattacini, 2017-11-09:
I think we can close this issue @apaikan @Nicogene since @arrenglover's concerns have been somehow taken into account.

For the outstanding points, I suggest you open a dedicated ticket on https://github.com/yarp/issues

closed 2020-12-08 by pattacini