robotology/community #256: iCub SEA legs - Effort and Velocity limits

fiorisi, 2017-10-12:
Which are the effort [Nm] and velocity [rad/s] limits for the iCub SEA legs joints?

fiorisi, 2017-10-12:
The issue collects the VELOCITY and the EFFORT joint limits for the iCub SEA legs (Series Elastic Actuator legs).
The SEA legs are generally assembled on all version of iCub 2.5 and on the first version of the Koroibot robot.

The following table recaps the effort and velocity limits for the complete leg. The maximum velocity is given for "zero torque", while the maximum effort corresponds to the maximum "stall torque".

|  JOINT NAME   |        NOMINAL       |     NOMINAL     |         PEAK         |      PEAK       |
|---------------|----------------------|-----------------|----------------------|-----------------|
|               | Max Velocity [rad/s] | Max Effort [Nm] | Max Velocity [rad/s] | Max Effort [Nm] |
| hip_pitch     |           2.5        |        25.4     |         5.1          |      55.5       |
| hip_roll      |           3.7        |        16.9     |         7.64         |       37        |
| hip_yaw       |           3.7        |        16.9     |         7.64         |       37        |
| knee          |           3.7        |        16.9     |         7.64         |       37        |
| ankle_pitch   |           3.7        |        16.9     |         7.64         |       37        |
| ankle_roll    |           3.7        |        16.9     |         7.64         |       37        |

The following plots depict the effort and velocity curves for the complete leg:
[image]

### Details on the calculations
The iCub 2.5 legs use the same electric motor and the same transmission for all the joints:
* electric motor : [MOOG][1] C2900584
* transmission: [HARMONIC DRIVE - CSD-17-100-2A-GR][2] (we use grease as the lubricant)
* motor drivers: [2FOC][3] 

The effort and velocity limits are computed taking into account the characteristics of all the components.  
In particular:
* the electric motor has a characteristic curve which defines the continuous operating region and has a peak stall torque limit
* the transmission has a maximum input speed limit, a rated output torque, and a peak output torque limit
* the electronics boards have a maximum operating voltage (can be different and charge-dependent if the robot is running on batteries), a nominal output current and a peak output current limit

By combining the "cumulative" effect of the limitations of each component, it is possible to construct the curves that define the continuous and peak operating region of the joints.

### Further details on the hip_pitch joint
The `hip_pitch` joint has an additional cable transmission stage after the harmonic drive. The motor side pulley has a diameter of 50 [mm], while the joint side pulley has a diameter of 75 [mm], consequently, the additional reduction ration is: `75/50 = `**`1.5`**  
The tendon is a [CarlStahl U7191517][4] steel cable. The minimum breaking load of the steel cable is 1590 [N]. The diameter of the pulley of the hip_pitch joint is 75 [mm], consequently, the torque limit due to the steel cable is: `1590*0.0375 = `**`59.625 [Nm]`**.

### Further details on velocity limits of the HD
Usually, transmissions have a maximum value for the input speed. Nevertheless, the HD that we are using for the leg joints have also an `average input speed` limit. The value of this limit can be found at page 10 of the [datasheet][2]. As defined at page 46, the `average input speed `is the "maximum average input speed". Since this value is not related to the torque, we use it to limit the continuous range.

[1]:https://www.moog.com
[2]:https://harmonicdrive.de/mage/media//catalog/category/2014_11_ED_1019659_CSD-2A.pdf
[3]:http://www.icub.org/images/brochures/iCub_controllers_flyer_web.pdf
[4]:https://www.carlstahl-technocables.de/fileadmin/files/katalog-carl-stahl-technocables.pdf

traversaro, 2017-10-12:
cc @InriaBrice that could be interested in this data.

traversaro, 2017-10-12:
> electric motor : MOOG C2900584

Caveat: in the past some models of the iCub v2.5 (SEA) legs used [Kollmorgen](http://www.kollmorgen.com) motors. This information used to be encoded in the robots configuration file in the `MotorType` parameters (see for example https://github.com/robotology/robots-configuration/blob/60b3d3b1ebb26de737b336ea2d22abc353ad16a9/iCubNancy01/hardware/mechanicals/right_leg-eb8-j0_3-mec.xml#L15 ) but I don't know if this information is updated.

closed 2020-12-15 by pattacini