robotology/community #257: Gazebo Cartesian controller movement issue

mpatacchiola, 2017-10-12:
Dear all,

I am having problem controlling the iCub model on Gazebo (both: iCub (fixed) and iCub with hands).
When I try to move the arm to a specific position the hand always moves in direction of the hip and it gets stuck there. After collapsing the arm does not move anymore. The same piece of code works perfectly on the standard iCub_SIM environment. 

At [this address](https://raw.githubusercontent.com/mpatacchiola/icub-demos/master/demoBall/issue/default_gzclient_camera(1)-2017-10-12T14_42_54.546548.jpg) you can see a screenshot of the issue. The following is the code I am using (slightly re-adapted from the [official tutorial](http://wiki.icub.org/brain/icub_cartesian_interface.html)): 

```
void trying_cartesian_controller(){
    Property option;
    option.put("device","cartesiancontrollerclient");
    option.put("remote","/icubSim/cartesianController/left_arm");
    option.put("local","/client/left_arm");
    PolyDriver clientCartCtrl(option);
    ICartesianControl *icart=NULL;
    if (clientCartCtrl.isValid()) clientCartCtrl.view(icart);
    //Declaring vectors and returning the current position
    Vector position_vector, orientation_vector;
    icart->getPose(position_vector, orientation_vector);
    std::cout << "x: " << position_vector[0] << std::endl;
    std::cout << "y: " << position_vector[1] << std::endl;
    std::cout << "z: " << position_vector[2] << std::endl;
    //New position
    Vector new_position_vector = position_vector; 
    new_position_vector[0]+=-0.1;
    Vector new_orientation_vector=orientation_vector;
    icart->goToPose(new_position_vector, new_orientation_vector);
    //SEnd and wait
    icart->goToPoseSync(new_position_vector,new_orientation_vector);
    bool done=false;
    while (!done) {
       icart->checkMotionDone(&done);
       Time::delay(0.04); 
    }
    clientCartCtrl.close(); //close the controller
}
```

I was considering a problem due to a different root frame. I tried moving the robot around the world and using small/large movements but the problem is always the same. Changing the arm gives the same issue. I tried using the rpc port and sending command directly from terminal but I got the same issue. Any idea?

Kind regard, Massimiliano

pattacini, 2017-10-13:
Hi @mpatacchiola 

I came across a very similar misbehavior while debugging https://github.com/robotology/icub-gazebo/issues/62.

Would you be willing to do your test again using this time the contexts I provided in [here][1]?
As described in that issue, the problem should be related to the `position-direct` vs. `velocity` control modes.

[1]: https://github.com/robotology/icub-gazebo/issues/62#issuecomment-336461717

pattacini, 2017-10-17:
Just as a quick suggestion, the following snippet
```c++
icart->goToPose(new_position_vector, new_orientation_vector);
icart->goToPoseSync(new_position_vector,new_orientation_vector);
bool done=false;
while (!done) {
    icart->checkMotionDone(&done);
    Time::delay(0.04); 
}
```
might be streamlined as:
```c++
icart->goToPoseSync(new_position_vector,new_orientation_vector);
icart->waitMotionDone(0.04);
```

mpatacchiola, 2017-10-17:
Dear @pattacini 

I had some improvements thanks to the new contexts you provide.
Now I can control the single joints from the rpc interface. 

However using the code posted before the arm moves differently. Now instead of going to the hip it goes to the chest. [Here](https://raw.githubusercontent.com/mpatacchiola/icub-demos/master/demoBall/issue/default_gzclient_camera(1)-2017-10-17T11_19_02.214079.jpg) is an image. 
The same issue remain, after moving to the chest the arm is stuck and I cannot move it anymore from code or rpc. I also tried to reduce the movement, increase the motion done delay to 0.1, and use the streamlined version you suggested. Nothing helped.

Some printing on terminal shows that the script stops inside the while loop or at `icart->waitMotionDone(0.04);`  in the streamlined version. It seems that the controller cannot reach the final position.

Which system configuration did you use to make it work? I can evaluate the hypothesis of downgrading Gazebo or YARP to make it work. My current configuration is:

1. Ubuntu 16.04
2. Gazebo 7.8.1
3. YARP and icub-main taken from the latest release on the repository

Thanks for your help.

Massimiliano

pattacini, 2017-10-17:
What are the resulting printouts of your code?

```c++
std::cout << "x: " << position_vector[0] << std::endl;
std::cout << "y: " << position_vector[1] << std::endl;
std::cout << "z: " << position_vector[2] << std::endl;
```

mpatacchiola, 2017-10-17:
When I run the script i ge these printoutst: 

```
x:0
y:0
z:0
```

However if I move the arm using rpc and then I run the script I get values which are different from zero.

Massimiliano

pattacini, 2017-10-17:
Ya, exactly as I was afraid of.
`(0,0,0)` is the origin located at the waist, where the robot is moving his hand to, actually.

What happens is that when your snippet opens up the driver, the ports underneath are not yet operational and the call to [`getPose`][1] fails. In this respect, check its return code, you'll find out that it's `false`. Thereby, as a quick workaround, you could do:

```c++
Vector position_vector, orientation_vector;
while (!icart->getPose(position_vector, orientation_vector))
    Time::delay(0.1);  // remember to release the quantum
 std::cout << "x: " << position_vector[0] << std::endl;
 std::cout << "y: " << position_vector[1] << std::endl;
 std::cout << "z: " << position_vector[2] << std::endl;
```

Then, concerning the reaching of the target location, it very depends on the quality of the low-level control, that is how the PID gains are tuned with respect to the task at hand as well as considering the perturbations (i.e. gravity). With this in mind, you could prevent the code from being stuck in an endless check with a "_simple_" timeout as it follows:

```c++
icart->waitMotionDone(0.04,3.0);
```

In this example, I used a timeout of `3.0` seconds that should be fine for a movement generated with a point-to-point trajectory time of, let's say, `1.0` second. 

[1]: https://github.com/robotology/icub-main/blob/master/src/libraries/icubmod/cartesianController/ClientCartesianController.cpp#L364-L393

mpatacchiola, 2017-10-18:
Great! The last suggestion seems to work. The `getPose` method now returns the correct location and the issue is gone. The context files are still necessary. In conclusion, if somebody else is having the same problem, the following solution worked form me. 

1. Extract the context files you can find [here](https://github.com/robotology/icub-gazebo/issues/62#issuecomment-336461717) in `~/.local/share`

2. Run the modified code:

```
void trying_cartesian_controller(){
    Property option;
    option.put("device","cartesiancontrollerclient");
    option.put("remote","/icubSim/cartesianController/left_arm");
    option.put("local","/client/left_arm");

    PolyDriver clientCartCtrl(option);
    ICartesianControl *icart=NULL;
    if (clientCartCtrl.isValid()) clientCartCtrl.view(icart);

    //Declaring vectors and returning the current position
    Vector position_vector, orientation_vector;
    while (!icart->getPose(position_vector, orientation_vector))
        Time::yield();  // remember to release the quantum
    std::cout << "x: " << position_vector[0] << std::endl;
    std::cout << "y: " << position_vector[1] << std::endl;
    std::cout << "z: " << position_vector[2] << std::endl;

    //New position
    Vector new_position_vector = position_vector; 
    new_position_vector[0]+=-0.1;
    Vector new_orientation_vector=orientation_vector;
    icart->goToPose(new_position_vector, new_orientation_vector);

    //Wait motion done
    icart->goToPoseSync(new_position_vector, new_orientation_vector);
    std::cout << "Waiting motion done..." << std::endl;
    icart->waitMotionDone(0.04, 3.0); //waiting for 3 seconds

    clientCartCtrl.close(); //close the controller
}
```

Thanks @pattacini for your help.

Massimiliano

closed 2020-12-08 by pattacini