robotology/community #258: YARP TCP vs. Fast_TCP

claudiofantacci, 2017-10-19:
Today, during a discussion about YARP network protocols, some of us discovered that the `TCP` is actually the **standard TCP + an ACK with YARP**, while the `Fast_TCP` is actually the **standard TCP** protocol.

Is it the case?
If that so, why aren't they called, respectively, `YARP_TCP` and `TCP`?
Finally, the `TCP` [documentation](http://www.yarp.it/carrier_config.html#carrier_config_tcp) and [`Fast_TCP`](http://www.yarp.it/carrier_config.html#carrier_config_fast_tcp) documentation seems to say something different.

cc @pattacini @randaz81 @vtikha @barbalberto @lornat75 @drdanz @Nicogene

pattacini, 2017-10-19:
I would love seeing the standard TCP (i.e. without YARP ack) becoming the default when we set up a `tcp` protocol for a YARP connection...

Nicogene, 2017-10-20:
According to @drdanz they are both yarp protocols, so we can't define them `yarp_tcp` and `tcp`. The only difference is that `tcp` carrier sends through the tcp socket the data in one way and the acknowledgement in the way back. If you need a synchronization between sender and receiver you have to use `tcp` otherwise `fast_tcp` is more efficient.
Set `fast_tcp` as default could be a good idea...

claudiofantacci, 2017-10-20:
Thanks @Nicogene and @drdanz for the explanation.

If there is nothing more to add we can close this Q&A.  
> Set fast_tcp as default could be a good idea...  
About this point, we may want to discuss it f2f, but anyway I reckon this is not the place to do it 😄

lornat75, 2017-10-23:
The idea behind TCP + ACK to be the default is that it simplifies communication for new users. Advanced users who better understand how communication works with YARP can use other protocols like udp or tcp_fast. Instead of changing default I believe we should improve the documentation and better highlight the differences.

lornat75, 2017-10-23:
If we want to clarify names we could deprecate the tcp protocol, call it tcp_ack and make this the default. In this way we do not change the default behavior (which could be catastrophic) but clarify the name of the protocol.

claudiofantacci, 2017-10-23:
> Instead of changing default I believe we should improve the documentation and better highlight the differences.

I think this can be done during one of the next IFI Days so that this point is made crystal clear to everyone 👍

pattacini, 2017-10-23:
Then, we need a huge manpower to update our xml-applications / user-code 😉

closed 2020-12-08 by pattacini