robotology/community #259: cannot connect to pc104

ems7, 2017-10-26:
After running the iCub at Heriot Watt university Edinburgh for several hours, I rebooted the system and am now unable to ssh to pc104. Any ideas what is likely to have caused this? The head was quite warm but nothing in the head looks or smells burnt. 
I have tried connecting through a router and also directly via the iCub's ethernet cable neither of which has worked.

Both ssh and ping show that the host is unreachable.

pattacini, 2017-10-26:
@mbrunettini @julijenv could you have a look at this?

mbrunettini, 2017-10-26:
Please try to replace USB memory inside the PC104. Instruction are [here ](http://wiki.icub.org/wiki/The_Linux_on_the_pc104)

ems7, 2017-10-26:
We have the PC104 with Atom processor, where is the usb stick on this model?

mbrunettini, 2017-10-26:
@julijenv : can you assist them in checking and replacing the USB?
Please do not overwrite the original USB, it can be still useful

ems7, 2017-10-27:
I got the iCub to boot.

`ifconfig`

indicates that enp3s0 is missing so there is no device with the ip address 10.0.0.1 

Looking in /var/log/syslog 
`cat syslog | tr -d '\000` | grep enp3s0`
I can see that at october 25 15:17 the enp3s0 device cannot be found.

In the log file i see that the network device is not initalised properly. Modprobe fails with an NVM checksum error.

The device shows up with lspci but not with lshw.

[image]

[image]

Thanks in advance, Eli

pattacini, 2017-11-03:
@mbrunettini you might have some comments on this.

mbrunettini, 2017-11-06:
Can you please post here the content of the following files (on icub-head) : 

- /home/icub/VERSION_iCubOS
- /etc/issue
- /etc/network/interfaces

As well as some pictures of the iCub head, taken on both sides (specially the USB connector area)
This could help us a lot in understanding your issue.
Thanks

ems7, 2017-11-08:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Unfortunately the iCub is no longer booting directly using the OS on the head. Using an ubuntu 16.04 livestick the iCub will boot but crashes as soon as the desktop is loaded or when trying to boot into the gnome terminal. As such, I can't get you the files you asked for. 
However, the software on the head hasn't been changed from when the iCub was shipped to us in february/march.

mbrunettini, 2017-11-08:
Using a generic live USB is not an option: the OS is a customized version of Ubuntu, so a generic live does not work.
So the screenshot you posted above is not from the original onboard OS, but from a live USB?
Cat you post here the error shown if you try to boot it with its original OS?
Thanks

ems7, 2017-11-08:
The screen shots from above were taken using the on board, original OS. I am unable to get the iCub to boot from a livestick.

mbrunettini, 2017-11-08:
Ok then, from what I can see from the screenshot, it seems an Hardware Failure .

pattacini, 2017-11-09:
@ems7 if the problem is actually a hardware failure, you should contact our maintenance group @robotology/icub-support-team.

pattacini, 2017-11-14:
@julijenv confirmed he's taking on this task.

closed 2020-12-15 by pattacini