robotology/community #262: iCub Nancy joints velocities limits

InriaBrice, 2017-11-14:
Hi,

So we need to know the joints velocities limits for our icub. Regarding this issue https://github.com/robotology/QA/issues/246 , can we admit that we have the same limits for our icub ? Atleast for the legs.

The whole body limits would be great for us.

Best regards,

@dogoepp @serena-ivaldi  @olivierrochel-inria @randaz81 @pattacini  @traversaro

traversaro, 2017-11-14:
cc @fiorisi

pattacini, 2017-11-14:
Hi @InriaBrice 

I quickly read #246 and there the topic is regarded with joint **position** limits, not velocities limits.

pattacini, 2017-11-14:
Perhaps you did mean #256

InriaBrice, 2017-11-14:
yeah my bad, it was 256

traversaro, 2017-11-14:
Regarding the legs: the mechanics of the legs of iCubNancy01 is indeed what we usually call iCub SEA legs (indeed, you should have a set of C-shaped compliant springs that you can mount on the robot). 
With respect to the applicability of the analysis in https://github.com/robotology/QA/issues/256 the only thing I am not sure is that the analysis assumes that the motors are the MOOG C2900584,  but I don't know which kind of motors are mounted on the iCubNancy01, see https://github.com/robotology/QA/issues/256#issuecomment-336091822 for more info.

fiorisi, 2017-11-14:
> but I don't know which kind of motors are mounted on the iCubNancy01

I can check with production.

randaz81, 2017-11-16:
Please also note that no velocity limiter is currently implemented at FW level.

fiorisi, 2017-11-16:
Next week we will inspect the list of components with our production department.

fiorisi, 2017-11-26:
The legs of iCubNancy01 should have the following motors:
* **Kollmorgen** (code to be checked: RBE01211-A) for `l/r_hip_roll` and `l/r_hip_pitch`
* **MOOG C2900574** for the `l/r_knee`
* **MOOG C2900525** for` l/r_hip_yaw`, `l/r_ankle_pitch`, `l/r_ankle_roll`

Having a look at the characteristics of the `MOOG C2900584` (used to compute the [joint characteristics](https://github.com/robotology/QA/issues/256)) and `MOOG C2900525` the two motors should be identical. Here attached the datasheet form our local repository.
[C2900584.pdf](https://github.com/robotology/QA/files/1503814/C2900584.pdf)
[C2900525.pdf](https://github.com/robotology/QA/files/1503815/C2900525.pdf)
 
Concerning the `MOOG C2900574` we have to check, but it is likely to be identical to the `MOOG C2900584` as well.

Can you check the configuration files as [suggested](https://github.com/robotology/QA/issues/256#issuecomment-336091822) by @traversaro?

pattacini, 2017-12-04:
@InriaBrice does @fiorisi's post respond to your question?

InriaBrice, 2017-12-11:
Sorry for the delay, i wasn't here. Yes he did thanks.

pattacini, 2017-12-11:
Perfect thanks @InriaBrice 👍

closed 2020-12-15 by pattacini