robotology/community #264: Recommended usage of Threads inside RFModules

kt10aan, 2017-12-02:
It is been quite a while since I have used a ``YARP Thread`` within an ``RFModule``.

I am writing a module now where I have to use threads again. Looking around to see if something has changed in the API, I found the ``RFModuleHelper`` and ``RFModuleThreadedHandler`` classes within ``RFModule``, which I hadn't noticed the last time.

I tried to find an example for the above but I could not. Could you point to to some existing code/module that it is using the above for starting/managing (YARP) threads?

Also, any "official" recommendation regarding good practices about where in the RFModule::config and RFModule::update I should put the ``Thread::beforeStart()``, ``Thread::init``, ``Thread::afterInit`` and ``Thread::start/run()`` methods?

claudiofantacci, 2017-12-02:
Hi @kt10aan 

> Looking around to see if something has changed in the API, I found the `RFModuleHelper` and `RFModuleThreadedHandler` classes within `RFModule,` which I hadn't noticed the last time.

The two above-mentioned classes are within the `RFModule` and are not part of the public API. Thus, they are not supposed to be used. The `RFModule` API is described [here](http://yarp.it/classyarp_1_1os_1_1RFModule.html) and some tutorials are available [here](http://www.yarp.it/yarp_resource_finder_tutorials.html).

> Also, any "official" recommendation regarding good practices about where in the `RFModule::config` and `RFModule::update` I should put the `Thread::beforeStart()`, `Thread::init`, `Thread::afterInit` and `Thread::start/run()` methods?

This is really something that depends upon your application. There are no "official" recommendation for this, but as far as my opinion is concerned, I would probably use the update function of the `RFModule` to check the state (controller) of the other `Thread` (controlled) and take proper action under specifc conditions. But please note again that I have no information about your use case.

drdanz, 2017-12-04:
Hi @kt10aan,

It depends on what you would like to do.

As @claudiofantacci wrote, the classes you mentioned are not supposed to be used.

In the normal use cases, the RFModule class should be started by calling either `runModule` (to block current thread until the module is stopped) or `runModuleThreaded` (to run it in a separate thread).

If you need an extra thread, there is no official recommendation, but my personal opinion is that you should not use `yarp::os::Thread` (de facto deprecated) but [`std::thread`](http://en.cppreference.com/w/cpp/thread/thread) instead.

kt10aan, 2017-12-04:
I want to have two threads that process incoming images from the iCub eyes, one for each eye. The way I had done it in the past, was to have a period of 0.0 for the ``RFModule::update`` (so to never update until interrupted) and start the two ``YARP::os::Thread`` in the ``RFModule::configure`` method.

But if you say that ``yarp::os::Thread is deprecated, I will switch to ``std::thread`` for future-proofing.

barbalberto, 2017-12-04:
Actually, a period of 0.0 means "run as fast as you can" and it is not so advisable. Use higher period instead. TheRFmodule checks every second if it has to be terminate, even if period is higher.

drdanz, 2017-12-04:
It's not "officially" deprecated, but since c++11 offers threads, so `yarp::os::Thread` is no longer that useful, and it is going to became a wrapper around `std::thread`, and at some point deprecated.

kt10aan, 2017-12-04:
@barbalberto, as fast as you can it is what I am aiming for.But I can see its drawbacks.

In any case, since I want to have two images, from the two eyes, processed concurrently and being sent concurrently (within reason), I thought of having two threads inside the module.

The other option is to have two modules, each for each eye, then send it to a third module to synchronize the outcome. But in that case I am worried about delays (image transfer is slow at 30fps@640x480, even with ``shmem``).

pattacini, 2017-12-04:
Just to complement what @barbalberto said about `RFModule::getPeriod()` returning `0.0`, in your case @kt10aan is not required since you have 2 threads for doing your processing.

When you return `0.0`, then in your `RFModule::updateModule()` you shall have some blocking calls to get in sync with external events, otherwise, you'll allow the RFModule to spin continuously wasting resources.

barbalberto, 2017-12-04:
@kt10aan You can achieve the best responsiveness if you spawn two thread, one for each eye, and operates whenever an image is received. This can be done by using a blocking read on the port receiving the images. As soon as you receive an image, you can process it, then sleep again untill nex image arrives.
For this you need a simple Thread, _not_ RateThread.

Please note that update function of RFModule has no relation with the spawned thread. I mean, there is no need of calling RF::update to have the threads working. They will do their job regardless of what RFModule is alive or sleeping.
So threads will work as fast as possible, but if RFmodule has no work to do (my guess from what I read, but I may be wrong), it can sleep.

kt10aan, 2017-12-04:
That was my intended use: Spawn two threads and use blocking reads on each port. 
I do want to use the update() function for listening for keystrokes, in order to change some parameters in the image processing.

So from what I can understand, a reasonable design would be to spawn two ``std::thread`` in the configure() method, and use the update() method with ``0.0`` return value for listening to keystrokes and passing back the changed values to the threads.

To be honest, in most simple (not threaded) ``YARP`` programs, I am using the ``update()`` with ``0.0``, and a ``while`` statement with blocking reads (essentially, just using the ``RFModule`` in order to get access to the ``RFModule`` goodies (command line parameters, ``ResourceFinder`` and config files, ``yarprun`` integration, etc).

lschilli, 2017-12-04:
You can also put the images in a (synchronized) queue when you receive them. Then you can (pre)process them with multiple worker threads at least for calculations that are independent of frame ordering. This will increase throughput if calculations take long but might increase latency.

kt10aan, 2017-12-04:
@lschilli, thanks! But maybe this will complicate threads and have discrepancies between the queue for the left and right eye if the frame rate of one or the other camera has a small dip?

lschilli, 2017-12-04:
Well, that depends on many factors, but if strict synchrony is required I would grab them pairwise or at least ensure they are linked at an early level. I think there is a specific grabber for this case. At lease I saw something called grabberDual: https://github.com/robotology/yarp/blob/master/src/libYARP_dev/src/devices/ServerGrabber/ServerGrabber.cpp

barbalberto, 2017-12-04:
> To be honest, in most simple (not threaded) YARP programs, I am using the update() with 0.0, and a while statement with blocking reads (essentially, just using the RFModule in order to get access to [...]

Ok, so the `RF::update` is actually doing some work with blocking calls, then it is fine. As usual, if you need to use mutexes, just be careful to have the protected zone as small as possible, and leave blocking calls outside mutexes.
Or you can spawn yet another thread for handling keystrokes :smiley:

pattacini, 2017-12-04:
@kt10aan detecting/using keystrokes is not a good way of interacting with a module. Rather, rely on RPC communication, hence on the use of `RFModule::respond()`.

Even better, prepare a thrift version of those RPC services 😉

lornat75, 2017-12-05:
With the adoption of C++11 some functionalities provided by YARP may become obsolete, yarp::os::Thread is one of those. However, there has been no official decision about removing them from YARP. Still, threading support in C++11 is quite rich, therefore it looks like a good idea not to remain stuck with what YARP offers.

kt10aan, 2017-12-06:
@lschilli, I saw the worker approach in the code of ``gazetool`` :)
@pattacini, thanks for the tip, I will use the ``respond()`` method :)

In the end I settled with two ``std::thread``s and an update period of ``0.0``.

pattacini, 2017-12-06:
Hi @kt10aan 

Why don't you report here the content of your `RFModule::updateModule()`?
A period of `0.0` is critical - as we said - therefore, it might be worth looking into it.

pattacini, 2017-12-13:
Closing for now.
Feel free to open it again if needed.

EliseiFrederic, 2018-01-17:
Hi,
I tend to use the .useCallback() and .onRead() capabilities of ports, to avoid the sort of threads that you describe here.
Any reason why noone commented on this ? Any reason why I should not use these ?
Thanks, Frédéric

pattacini, 2018-01-17:
@EliseiFrederic an approach resorting to port callbacks is perfectly viable 👍

closed 2020-12-15 by pattacini