robotology/community #265: New iCub LIVE OS release

mbrunettini, 2017-12-06:
Hi,
We are happy to announce **a new release of iCub LIVE OS** for the iCub PC104:
  **iCub LIVE OS version 7.9**
This version is based on the latest Debian stable release, codename “Stretch”.

You can download it here:
  [](http://www.icub.org/download/pc104-images/)

Please follow the instructions on iCub wiki:
 []( http://wiki.icub.org/wiki/The_Linux_on_the_pc104)

gsaponaro, 2017-12-07:
Hi @mbrunettini, thanks for this release.

I report below one comment (by @EliseiFrederic in https://github.com/robotology/icub-support/issues/494#issuecomment-349603270), in case other people find it useful:
> I had to correct ONE ERROR in the new distro :
> in /etc/rc/iCub.d/mounts.lists
> soft,rerty=1,timeo=60,_netdev,auto
> should have "retry" instead of "rerty"

EliseiFrederic, 2017-12-07:
This might be useful too : 
http://wiki.icub.org/wiki/Compilation_on_the_pc104#Getting_YARP.2C_iCub_sources_and_Robots_configurations
but beware the error in the "Configuring your robot" part : 
"Make sure that CMAKE_INSTALL_PREFIX points to $ICUBContrib_DIR and finally do: " should read
"the value of $ICUBcontrib_DIR" , not the $variable and with a lower caps 'c'

pattacini, 2017-12-13:
@mbrunettini can we close this?
Do you have any final comments about @gsaponaro and @EliseiFrederic posts?

EliseiFrederic, 2017-12-15:
Still trying to have iCubGrenoble01 to start a working cfw2can...

pattacini, 2017-12-19:
Given https://github.com/robotology/icub-support/issues/583#issuecomment-352419243, I think a new announcement will be out soon.

Closing then.

closed 2020-12-12 by pattacini