robotology/community #266: usb system key unplugged

pfdominey, 2017-12-08:
The USB connector for the USB disk for the iCub pc104 came unplugged, and I simply do not know where it plugs back.  in the attached photo, the USB key is hanging down from the red arrow (so we cant see it but it is there), the connector that came uplugged is marked by thewhite arrow, and I think it plugs in somewhere near the yellow arrow but not sure.

can you please help me to find where to re-plug this?  should be quite simple for someone who knows :)
[iCubUSBConnection.pdf](https://github.com/robotology/QA/files/1542271/iCubUSBConnection.pdf)

pattacini, 2017-12-08:
Hi @pfdominey 

The picture below should give you the heads-up

[image]

Thanks @vtikha for the photo.

pfdominey, 2017-12-08:
Perfect!  Thanks very much

pfdominey, 2017-12-08:
Beautiful photo! Thanks Ugo

-- 
Peter Ford Dominey PhD, CNRS Research Director
Inserm U846
Stem Cell and Brain Research Institute
Integrative Neuroscience & Robotics
18, avenue du doyen Jean Lépine
69675 Bron cedex
France
Tél : +33-4 72 91 34 84
Fax : +33-4 72 91 34 61
http://www.sbri.fr/teams/human-and-robot-interactive-cognitive-systems.html
http://www.sbri.fr

closed 2020-12-15 by pattacini