robotology/community #270: How to change control mode in python

nigno17, 2018-01-15:
Hello everyone,

How can I change back the joints control mode in position after using the cartesian control in python? 

I need to first control the robot in position mode and then use the cartesian controller to control in task space. This process needs to be repeated in a loop.

Thanks in advance,
Nino

nigno17, 2018-01-15:
I used the setControlMode function in the following way:

____________________________________
props = yarp.Property()
props.put("device","remote_controlboard")
props.put("local","/client/right_arm")
props.put("remote","/icubSim/right_arm")

armDriver = yarp.PolyDriver(props)
iPos = armDriver.viewIPositionControl()
jnts=iPos.getAxes()
...
...
iMode = armDriver.viewIControlMode()

vocabPos = yarp.Vocab()
vocabPos.encode('pos')
for i in range(jnts):
    iMode.setControlMode(i, vocabPos)
______________________________________

When I run my code I receive this error:

Traceback (most recent call last):
  File "icubTest.py", line 171, in <module>
    iMode.setControlMode(i, vocabPos)
  File "/usr/local/lib/python2.7/dist-packages/yarp.py", line 10687, in <lambda>
    __getattr__ = lambda self, name: _swig_getattr(self, IControlMode, name)
  File "/usr/local/lib/python2.7/dist-packages/yarp.py", line 74, in _swig_getattr
    return _swig_getattr_nondynamic(self, class_type, name, 0)
  File "/usr/local/lib/python2.7/dist-packages/yarp.py", line 69, in _swig_getattr_nondynamic
    return object.__getattr__(self, name)
AttributeError: type object 'object' has no attribute '__getattr__'

Not sure if I'm doing it in the proper way. It seams that the bindings for IControlMode do not exist yet.

Thanks,
Nino

pattacini, 2018-01-17:
I'm not a Python user, so anyone eager to help here, or with the right heads-up on this?

jgvictores, 2018-02-02:
Hi @nigno17 @pattacini !!

Using the old `IControlMode` interface, you should still be able to use:
```python
props = yarp.Property()
props.put("device","remote_controlboard")
props.put("local","/client/right_arm")
props.put("remote","/icubSim/right_arm")

armDriver = yarp.PolyDriver(props)
iPos = armDriver.viewIPositionControl()
jnts=iPos.getAxes()
...
...
iMode = armDriver.viewIControlMode()
for j in range(1,jnts):
    iMode.setPositionMode(j)
```

Bindings for `IControlMode2` are still not in `master`, but in `devel` as of https://github.com/robotology/yarp/commit/554b7f64418a1949a2cce053d099457a11899807. I'll brew up an example if I have time and remember. :-)

nigno17, 2018-02-05:
Thank a lot. 
The old IControlMode interface works perfectly.

closed 2020-12-08 by pattacini