robotology/community #271: Did "yarp terminate" change its behaviour ?

EliseiFrederic, 2018-01-17:
Hi,

With our previous yarp version (2.3.64.8), we used yarp terminate a lot from the command line. For example to perform
yarp terminate /yarphear 
and that worked gracefully, whatever the targeted port.
So did it with our programs that implemented interruptModule() from RFMmodule.
It seems (our new version is 2.3.70.1) that it doesn't work anymore... Did the policy change ? 

Thanks in advance for the help.

pattacini, 2018-01-17:
@Nicogene do you mind checking this?

Nicogene, 2018-01-18:
Hi!
I didn't know about the existence of this command, I never used it but I just tried to run: 
```
$ yarpdev --device test_grabber
```
and 
```
$ yarp terminate /grabber 
```
and finally 
```
$ yarp terminate /grabber
```
it closed the `test_grabber`.
So took a look to the code of `yarphear` and I find out that internally it is actually a `device` like the `test_grabber`. The devices in `YARP` have the "quit" port, you can connect to it to close the device.

**CONCLUSION**
`yarp terminate` connect to the "quit" port of the device and tell him to shut down, `yarphear` is a device (in particular `portaudio`) so `yarp terminate` works smoothly with it.
With a generic `RFModule` I'm afraid that it doesn't work, but I think it never worked.

If you have further doubts or questions don't hesitate to ask!

EliseiFrederic, 2018-01-18:
Thanks for checking 👍  What you observe with your yarpdev is what I expect too...
That's what I get on my previous yarp install. Or if I have the yarp server running on my previous install.
But that's not what I get with my fresh install (updated to 2.3.70.2) !  

With the new version : 
$ yarpdev --device test_grabber
creates two visible ports on my yarp server : 
/grabber
and 
/test_grabber/quit
yarp terminate /grabber does nothing
yarp terminate /test_grabber does nothing
yarp terminate /test_grabber/quit does the expected thing though ! 

This is not only a device problem... On the old version, I can do
yarp write /test
and get it closing nicely with a yarp terminate /test
(there exists no /test/quit for this example...)

EliseiFrederic, 2018-01-31:
Thanks for looking to these details. We are using Linux (Jessie).
Our yarp is built from the sources.

If I send "TERM" signal to the "yarp write /test" process, it reacts as Interrupting, as expected from a "yarp terminate" command.

libACE-6.2.8 is available on the laptop, but I don't know if it's used to build yarp. Here are our flags : 
 ALLOW_IDL_GENERATION             OFF                                                                                                                    
 CMAKE_BUILD_TYPE                 Release                                                                                                                
 CMAKE_INSTALL_PREFIX             /usr/local                                                                                                             
 CREATE_DEVICE_LIBRARY_MODULES    OFF                                                                                                                    
 CREATE_GUIS                      ON                                                                                                                     
 CREATE_LIB_MATH                  ON                                                                                                                     
 CREATE_OPTIONAL_CARRIERS         ON                                                                                                                     
 CREATE_YARPBATTERYGUI            OFF                                                                                                                    
 CREATE_YARPDATADUMPER            ON                                                                                                                     
 CREATE_YARPDATAPLAYER            ON                                                                                                                     
 CREATE_YARPLASERSCANNERGUI       OFF                                                                                                                    
 CREATE_YARPLOGGER                ON                                                                                                                     
 CREATE_YARPMANAGER               ON                                                                                                                     
 CREATE_YARPMANAGER_CONSOLE       ON                                                                                                                     
 CREATE_YARPMOTORGUI              ON                                                                                                                     
 CREATE_YARPROBOTINTERFACE        ON                                                                                                                     
 CREATE_YARPSCOPE                 ON                                                                                                                     
 CREATE_YARPVIEW                  ON                                                                                                                     
 CUDA_BUILD_CUBIN                 OFF                                                                                                                    
 CUDA_BUILD_EMULATION             OFF                                                                                                                    
 CUDA_HOST_COMPILER               /usr/bin/cc                                                                                                            
 CUDA_SDK_ROOT_DIR                CUDA_SDK_ROOT_DIR-NOTFOUND                                                                                             
 CUDA_SEPARABLE_COMPILATION       OFF                                                                                                                    
 CUDA_TOOLKIT_ROOT_DIR            CUDA_TOOLKIT_ROOT_DIR-NOTFOUND                                                                                         
 CUDA_VERBOSE_BUILD               OFF                                                                                                                    
 DOXYGEN_PLANTUM_JAR              DOXYGEN_PLANTUM_JAR-NOTFOUND                                                                                           
 ENABLE_YARPRUN_LOG               OFF                                                                                                                    
 ENABLE_yarpcar_bayer             ON                                                                                                                     
 ENABLE_yarpcar_depthimage        OFF                                                                                                                    
 ENABLE_yarpcar_human             OFF                                                                                                                    
 ENABLE_yarpcar_mjpeg             OFF                                                                                                                    
 ENABLE_yarpcar_mpi               OFF                                                                                                                    
 ENABLE_yarpcar_mpibcast          OFF                                                                                                                    
[... cut ...]
 YARP_COMPILE_BINDINGS            OFF                                                                                                                    
 YCM_DIR                          YCM_DIR-NOTFOUND                                                                                                       
 wxWidgets_CONFIG_EXECUTABLE      wxWidgets_CONFIG_EXECUTABLE-NOTFOUND                                                                                   
 wxWidgets_wxrc_EXECUTABLE        wxWidgets_wxrc_EXECUTABLE-NOTFOUND

drdanz, 2018-02-09:
The 'i' argument was working on linux only and it was considered harmful, therefore it was enabled only when YARP is built in Debug mode since yarp 2.3.70, see [here](https://github.com/robotology/yarp/blob/master/src/libYARP_OS/src/PortCoreInputUnit.cpp#L332-L339) and robotology/yarp@3f9ef5d1ae0861b15c56c75c9707c70281f779a9

As a consequence, the yarp terminate command now works only on the "/quit" ports, unless YARP is built in Debug mode.

EliseiFrederic, 2018-02-09:
Thanks for pointing this change ! No more mystery now

closed 2020-12-15 by pattacini