robotology/community #272: Direct position mode : [ERROR] setPidReferenceRaw not yet implemented for CanBusMotionControl

EliseiFrederic, 2018-01-17:
Hi,

Our homebrew software to drive the talking head (jaw, lips...) does not work anymore.
We now have yarprobotinterface that reports a stream of
[ERROR] setPidReferenceRaw not yet implemented for CanBusMotionControl

The code was built from generalPlayer (now trajectoryPlayer), using
setReference() on a IPidControl object obtained as a view from a PolyDriver object.

Is it an issue related to the talking head, or an API change for direct position mode ?

Is there a way this can work again ? Should we switch to a IPositionDirectRaw object and call setPositionRaw() or will it fail for the same reason ?
Where should I get the IPositionDirectRaw object from ?

Thanks in advance for the help,
Frédéric

pattacini, 2018-01-17:
Hi @EliseiFrederic 

The correct way is to use the service [`yarp::dev::IPositionDirect::setPosition()`][1] of the `IPositionDirect` interface.

I might be missing the reasons why you went for `yarp::dev::IPidControl`, but I believe this way is deprecated (yet perhaps still functional) and thus my warm advice is to upgrade your code accordingly.

cc @randaz81 

[1]: http://www.yarp.it/classyarp_1_1dev_1_1IPositionDirect.html#ae80f52b80c498e14094697909584671a

randaz81, 2018-01-18:
I confirm that that method has been deprecated some time ago. You can now use `yarp::dev::IPositionDirect::setPosition() ` instead. You should check that the joint has been previously set in position direct mode by `yarp::dev::setControlMode(j, VOCAB_CM_POSITION_DIRECT)`

EliseiFrederic, 2018-01-18:
Thanks for the hint. With the suggested API, everything works again 👍

closed 2020-12-08 by pattacini