robotology/community #276: icubSim: is it possible to detect joints effort due to object collision?

valeriosperati, 2018-02-13:
Please, I've the following question:

Setup: 
-icubSim in front of a table with multiple objects, using cartesian controller to reach and move them.
-some objects are movable, other are static.

Problem:
When Icub tries to reach and move a static object, it keeps on the arm action, trying to reach the (impossible) final target position. This ends with simulated icub break.

Is there a way to detect a physical stress on simulated joints, in order to give up with the current action?
Thanks, Valerio

randaz81, 2018-02-13:
I'm not sure about iCubSim, maybe someone else can help you on this.
But for sure, this information is available if you use the iCub model for Gazebo.

randaz81, 2018-02-13:
relevant repos and installation instructions for icub simulation on gazebo:
- https://github.com/robotology/gazebo-yarp-plugins
- https://github.com/robotology/icub-gazebo

pattacini, 2018-02-19:
Hi @valeriosperati 
I don't believe `iCub_SIM` is comparable with its Gazebo counterpart in terms of handling of the dynamics, as it is the case here. Thus, the suggestion is to try to switch to Gazebo as @randaz81 already pointed out.

You could think of monitoring the difference between the commands the Cartesian Controller sends to the robot and the encoders feedback and attempt to figure out a possible strategy to detect collisions. However, I'm afraid that this road is not promising in that I argue it will be difficult to set up suitable thresholds that work in normal operations and at the same time will allow catching collisions quickly.

valeriosperati, 2018-02-19:
Ok, thanks a lot for suggestions.

closed 2020-12-08 by pattacini