robotology/community #281: Running the "Getting accustomed with motor interfaces" tutorial?

stango12, 2018-02-28:
Hello,

I was trying to run the program from the tutorial "Getting accustomed with motor interfaces" but was faced with an error pasted here:

> $./motion --robot icubSim
> [ERROR]Cannot find "remove_controlboard" plugin (not built in, and no .ini file found for it)
> [ERROR]Check that YARP_DATA_DIRS leads to at least one directory with plugins/remove_controlboard.ini or share/yarp/plugins/remove_controlboard.ini in it
> yarpdev: ***ERROR*** could not find device <remove_controlboard>
> Device not available. Here are the known devices:
> Device "test_motor", C++ class yarp::dev::TestMotor, wrapped by "controlboard"
> Device "remote_grabber", C++ class yarp::dev::RemoteFrameGrabber, wrapped by "grabber"
> Device "grabber", C++ class yarp::dev::ServerFrameGrabber, is a network wrapper.
> Device "sound_grabber", C++ class yarp::dev::ServerSoundGrabber, is a network wrapper.
> Device "pipe", C++ class yarp::dev::DevicePipe, has no network wrapper
> Device "group", C++ class yarp::dev::DeviceGroup, has no network wrapper
> Device "remote_controlboard", C++ class yarp::dev::RemoteControlBoard, wrapped by "controlboardwrapper2"
> Device "inertial", C++ class yarp::dev::ServerInertial, is a network wrapper.
> Device "controlboardwrapper2", C++ class yarp::dev::ControlBoardWrapper2, is a network wrapper.
> Device "batteryClient", C++ class batteryClient, has no network wrapper
> Device "analogsensorclient", C++ class AnalogSensorClient, has no network wrapper
> Device "analogServer", C++ class yarp::dev::AnalogWrapper, is a network wrapper.
> Device "virtualAnalogServer", C++ class yarp::dev::VirtualAnalogWrapper, is a network wrapper.
> Device "batteryWrapper", C++ class yarp::dev::BatteryWrapper, is a network wrapper.
> Device "Rangefinder2DWrapper", C++ class yarp::dev::Rangefinder2Dwrapper, is a network wrapper.
> Device "RGBDSensorWrapper", C++ class yarp::dev::RGBDSensorWrapper, is a network wrapper.
> Device "RGBDSensorClient", C++ class yarp::dev::RGBDSensorClient, is a network wrapper.
> Device "controlboardremapper", C++ class yarp::dev::ControlBoardRemapper, wrapped by "controlboardwrapper2"
> Device "remotecontrolboardremapper", C++ class yarp::dev::RemoteControlBoardRemapper, wrapped by "controlboardwrapper2"
> Device "robotDescriptionServer", C++ class RobotDescriptionServer, has no network wrapper
> Device "robotDescriptionClient", C++ class RobotDescriptionClient, has no network wrapper
> Device "grabberDual", C++ class yarp::dev::ServerGrabber, is a network wrapper.
> Device "JoypadControlServer", C++ class yarp::dev::JoypadControlServer, is a network wrapper.
> Device "JoypadControlClient", C++ class yarp::dev::JoypadControlClient, is a network wrapper.
> Device "transformServer", C++ class yarp::dev::FrameTransformServer, is a network wrapper.
> Device "transformClient", C++ class transformClient, has no network wrapper
> Device "Rangefinder2DClient", C++ class Rangefinder2DClient, has no network wrapper
> Device "navigation2DClient", C++ class navigation2DClient, has no network wrapper
> Device "localization2DClient", C++ class localization2DClient, has no network wrapper
> Device "map2DClient", C++ class yarp::dev::Map2DClient, has no network wrapper
> Device "map2DServer", C++ class yarp::dev::map2DServer, is a network wrapper.
> Device "cartesiancontrollerclient", available on request (found in /home/osboxes/icub-main/build/lib/iCub/cartesiancontrollerclient.so library), is a network wrapper.
> Device "cartesiancontrollerserver", available on request (found in /home/osboxes/icub-main/build/lib/iCub/cartesiancontrollerserver.so library), wrapped by "cartesiancontrollerclient".
> Device "gazecontrollerclient", available on request (found in /home/osboxes/icub-main/build/lib/iCub/gazecontrollerclient.so library), is a network wrapper.

I've tested if I installed yarp and icub correctly by running "yarpserver" and "iCub_SIM" and they both work. Any help would be appreciated. Thanks!

traversaro, 2018-03-01:
Hi @stango12 , the error message prints: 
> Cannot find "**remove_controlboard**" 

Are you sure that you don't have a typo in your source code with remo**v**e_controlboard instead of  remo**t**e_controlboard ?

stango12, 2018-03-01:
You were completely correct. I just a typo in my source code that I never noticed. Thank you so much for the help! Can't believe I missed that...

closed 2020-12-15 by pattacini