robotology/community #283: Yarp bindings  python

robertorovella91, 2018-04-05:
Hi, 
We are trying to use python in order to control the icub, but after installing with python3.5 the libraries, following the instructions in the README, the example files in yarp/bindings/python/examples doesn't start and this error occurs

" AttributeError: module 'yarp' has no attribute 'Network' "
How we can solve the problem? 
Thanks in advance,
Roberto

pattacini, 2018-04-09:
Any Python developer eager to help on this?
@robotology/all-robotology-developers

traversaro, 2018-04-09:
Hi @robertorovella91, what is the content of your  [`PYTHONPATH`](https://docs.python.org/3/using/cmdline.html#envvar-PYTHONPATH) environmental variable? Which OS are you using?

robertorovella91, 2018-04-10:
Hi @traversaro 
The problem was that we had not setted PYTHONPATH with the folder containing yarp.so file. Now we have fixed and solved the problem.
Thank you.

ZiaMohiUDin, 2019-09-27:
I am in same problem, (module 'yarp' has no attribute 'Network')
I added to path, but not solved,
[image]

ZiaMohiUDin, 2019-09-27:
expecting soon response,.. windows 10

alecive, 2019-09-27:
Me too, thanks, I will add another message in 10 minutes in case you haven't seen this.

alecive, 2019-09-27:
Ops sorry I meant 10 seconds.. here's my second message.

ZiaMohiUDin, 2019-09-28:
@robertorovella91
Hi I no folder containing yarp.so file

closed 2020-12-15 by pattacini