robotology/community #285: Accidentally changed the owner of binary files from root to icub (username) in server laptop

tasbolat1, 2018-04-06:
Hello everyone,

During the work with iCub laptop (server), I  ran _sudo chown -R /_ command, which makes icub (username) to be the owner of all possible binary files, including sudo. Hence, programs (which needs root privileges) cannot be run anymore. Specifically, as a result
- cannot connect pc104 of icub
- can setup neither wire nor wireless connections
- cannot get intro RecoveryMode

When I run the _sudo_ command, I get the following error:
_/usr/bin/sudo must be owned by uid 0_

The iCub Laptop (server) came with iCub robot with pre-installed binaries set-up to work with robot.
After a long time search in the internet, I haven't found any solution for the problem. Could you help me with this issue?

Thanks in advance,
Tasbolat

caicaixia, 2018-04-06:
:(

lornat75, 2018-04-06:
can you try logging in as root? root should be able to re-assign correct ownership to files.

tasbolat1, 2018-04-06:
@lornat75 Yes, I've tried, but no success.

To log in as a root from a terminal I need to have _sudo_ privileges. However, I cannot run _sudo_. I also cannot boot into recovery mode.

mbrunettini, 2018-04-06:
Regarding the error with sudo command, it must be owned by root and have the suid bit setted
`chown root:root /usr/bin/sudo && chmod 4755 /usr/bin/sudo`
see [this link](https://askubuntu.com/questions/452860/usr-bin-sudo-must-be-owned-by-uid-0-and-have-the-setuid-bit-set)

tasbolat1, 2018-04-06:
@mbrunettini 
I've checked the link, but problem is, again, I cannot log in as root. So, that solution doesn't work for this case.

mbrunettini, 2018-04-06:
can you login as icub?

tasbolat1, 2018-04-06:
yes

mbrunettini, 2018-04-06:
did you tried to issue the above command as icub, then?

tasbolat1, 2018-04-06:
Yes, I've tried. Have the following error:
_Permission denied_

mbrunettini, 2018-04-06:
`ls -la /usr/bin/sudo` 
?

tasbolat1, 2018-04-06:
`-rwxr-xr-x 1 icub root 136808 lug 4 2017 /usr/bin/sudo`

mbrunettini, 2018-04-06:
you used this laptop with the robot?
Only this one or other machines?

tasbolat1, 2018-04-06:
Yes, I used the laptop with the robot. It’s only machine working as a server

drdanz, 2018-04-06:
You could also try logging in with a live distro (for example [SystemRescueCd](http://www.system-rescue-cd.org/)), mounting the hard drive and changing the ownership of the files from there...

mbrunettini, 2018-04-09:
If it is already installed, of course not.

tasbolat1, 2018-04-09:
I have a concern in the following part:

> git clone https://github.com/robotology/yarp.git

Should I do it as `root` or `icub`?

mbrunettini, 2018-04-09:
icub
There is no `sudo`

tasbolat1, 2018-04-10:
We followed the instruction given and could connect to the robot. We could also run `yarprobotinterface`.
However, when we run `yarpmotorgui`, we get the following error:
[image]

mbrunettini, 2018-04-10:
It is a different issue, i propose to close this one and to open a new one with a different topic

tasbolat1, 2018-04-10:
Let me clear one thing:
We reinstall everything in the icub@icubsrv, but we didn't re-install or change anything inside the pc104.
So that means we downloaded and reinstalled all the codes in the server laptop (e.g. yarp, icub-main and so on) in /exports/code.  So I am wondering if the version of those code in pc104 is not match or link to this server laptop anymore.  Then it causes this problem?

My point is how to synchronize or map those source codes between the pc104 and server laptop?

barbalberto, 2018-04-10:
Yes, you have to update YARP on the robot

tasbolat1, 2018-04-10:
@barbalberto How shall we do? Do we need to get inside of the pc104 and git pull , then re-build it.
Or we can somehow to map from the server laptop?

barbalberto, 2018-04-10:
You have to go inside the pc104
```
ssh on pc104
cd $YARP_DIR
git pull
cmake .. 
make -j2
```
then do the same on ICUB_DIR

barbalberto, 2018-04-10:
And other folders you have in /usr/local/src/robot like icub-firmware-shared, in the order shown in the page http://wiki.icub.org/wiki/Compilation_on_the_pc104 from section compile yarp onward

mbrunettini, 2018-04-10:
since the issue about the laptop is solved, i close this

barbalberto, 2018-04-10:
Ok, you need also to update firmware on the boards.
Can you please tell us your YARP_ROBOT_NAME? Then we'll open a separated issue for that.

tasbolat1, 2018-04-10:
iCubSingpapore01

barbalberto, 2018-04-10:
I created issue here https://github.com/robotology/icub-support/issues/626

closed 2020-12-15 by pattacini