robotology/community #287: How to uninstall YARP from Windows - binary release

arrenglover, 2018-04-18:
Hi Guys,

I have installed YARP on windows using the binaries available here: http://www.yarp.it/download.html

I would now like to install YARP on Windows from sources, however I would first like to uninstall YARP to make sure I don't end up with version mis-match. I can't find any information about how to uninstall the binaries. 

Is there an easy way to do it? Is it a case of deleting the files and removing any YARP related paths from the system path?

lornat75, 2018-04-19:
anyway, you can uninstall them manually by removing all files from the installation tree, clean the environment from variables (PATH variable and *_DIR variables), there are also a few registry keys you can manually remove (search for YARP and iCub)

arrenglover, 2018-04-23:
Thanks for your follow up response @lornat75

closed 2020-12-08 by pattacini