robotology/community #289: keyboard as device in yarp

mhubii, 2018-04-20:
Hi and thanks for taking the time,

is there a way to use a keyboard as a device in yarp, such that the robot can for example move by pressing w, a, s, and d? I would need some kind of interface that reads the keyboard input to a port. Is there a way to go?

Martin

mhubii, 2018-04-21:
Okay sry I am pretty new to yarp. To solve this problem, one can implement a usual keyboard input via ncurses on linux and stream the commands using ports. If someone else is interested in how to implement this, check out my heicub_walking repository, it will be uploaded there soon.

Martin

randaz81, 2018-04-23:
wait, you don't need to develop anything new! You can use the joystick driver already provided in YARP via SDL. The name of the device is SDLJoypad. 

Before starting it, just launch your preferred keyboard to joypad emulator.

drdanz, 2018-05-02:
@randaz81 this is quite interesting and useful, is it documented somewhere?

closed 2020-12-15 by pattacini