robotology/community #291: Linear constant velocity of the icub hand

tasbolat1, 2018-05-16:
Hello, everyone!
I want the hand of icub to move with constant velocity along `y` axis. To do so, I used Cartesian Control and with the following settings:
```

xdot[0]=0.0; xdot[1]=-0.003; xdot[2]=0.0;
odot = 0.0;
icart->setTaskVelocities(xdot, odot);
```

where `icart` is `ICartesianControl`. I haven't set any other configurations (no trajectory time). And, asked the icub hand to move to desired position like this:

icart->goToPose(xd,od);

where hand moves form  (-0.3, 0, 0.1) to (-0.3. -0.3, 0.1). However, the icub simulation result doesn't correspond to what I asked. Here are figures:
Motion along `y` axis:
[image]

Motion along `x` axis
[image]

Motion along `z` axis
[image]

about figures:
 - we expect that there's no change (at least very small) in `x` and `z` axes. However, they change more than 4 cm and 1 cm respectively,  which, I think it's too big.
- in the first figure (along `y` axis), the velocity hasn't been set as I set using `setTaskVelocities`.

So, my questions are:
- Where I'm doing wrong?
- Is it, at the end of the day, feasible with Icub hand to have a linear motion with constant speed?
- If so, please suggest ways to do so.

Thanks, Tasbolat

xEnVrE, 2018-05-16:
Hi @tasbolat1,

it is indeed possible to have the iCub hand to follow a constant linear velocity trajectory. Let me clarify some points before getting to a possible solution to the problem.

As far as I understand you are trying to accomplish this task by first issuing a `setTaskVelocities` command and then a `goToPose` command. In this case the first command would be **overridden** by the second and the Cartesian Controller will essentially perform a point to point control moving the palm of the hand from the initial position ` (-0.3, 0, 0.1)` to the final position ` (-0.3, -0.3, 0.1)` using a Minimum-Jerk trajectory. This kind of trajectory has several desirable properties but, as you can find out from the figures that you have attached, it does not correspond to a straight line from the initial to the final position as you expected.

In order to achieve your goal you should issue the `setTaskVelocities` command alone. When such a command is issued the Controller **integrates** internally the commanded velocity and produces a reference trajectory whose position and orientation are tracked as normally happens when you issue a `goToPose` command - you can image this as issuing multiple times the `goToPose` command with intermediate poses along the desired trajectory that is specified in terms of its  **instantaneous** velocity using `setTaskVelocities`.

A very important parameter to be set in order to shape the resulting trajectory is the trajectory time. The trajectory time is not just the time required to move the end-effector from the initial pose to a given final pose. In fact it determines the reactivity  of the Controller, i.e. **lower** values of the trajectory time determines **faster** movements of the robot and vice-versa. This also applies to the `setTaskVelocities` command since, as I explained, it essentially produces several intermediate poses that will be tracked according to the specified trajectory time.

I would suggest you to read this [issue](https://github.com/robotology/QA/issues/208) that provides useful insights on how to tune the trajectory time when performing control with `setTaskVelocities` and also some [example code](https://gist.github.com/pattacini/41355442915d11b7b77eb46770f3df52).

Finally please note that, although you are not setting a trajectory time, a default value of `2.0s` is set within the Controller.

tasbolat1, 2018-05-22:
@xEnVrE , thank you very much directing me to the place we I could find useful informations!

Regarding the use of `setTaskVelocities`, I have some questions.

The problem statement (for now, everything is done in simulation):
I still need to have a constant linear velocity between `(-0.3, 0, 0.1)` and `(-0.3, -0.25, 0.1)`. The displacement has been shrunk a little bit to ensure that those points are reachable. Using `setTaskVelocities`, I can get the following result for different trajectory time:

Parameters:
`Update_period = 0.02 [s]`
`Velocity = -0.03 [m/s]` along y-axis
`Traj_time == 0.1 s:`
[image]

`Traj_time == 0.4 s:`
[image]

`Traj_time == 0.7 s:`
[image]

`Traj_time == 1 s:`
[image]

About figures:
`red = x, green = y, blue = z`
The velocity profile data has been recorded by `getTaskVelocities` during each iteration of loop.

The questions are:
1. Can we have the more smooth velocity profile than given in the figures? If so, which parameter should be tuned?
2. When the hand reaches the final point, 0 velocity is set using setTaskVelocities. However, as seen in figures, it sharply stops the motion. This case may damage real robot (it's my guess).
3. In real robot, how much external friction will have an effect?

Thanks,
Tasbolat

tasbolat1, 2018-06-07:
Thanks @pattacini  and @xEnVrE for the help. I could manage to achieve constant linear velocity to the end-effector using above explained methods in real robot. Of course, the velocity is not absolutely constant, but it's acceptable in my task.
Thanks again,
Tasbolat

pattacini, 2018-06-07:
Happy to have helped you out 👍

closed 2020-12-08 by pattacini