robotology/community #292: dragonfly2 with Bayer: Sharpness, hue, gamma

Tobias-Fischer, 2018-05-18:
Hi,
We recently started using the iCub cameras with Bayer decoding. We can view the images in colour etc, and the Bayer carrier is enabled.
However, in the frameGrabberGui, it doesn't allow us to set the sharpness / hue / gamma / saturation anymore. We found that we can set the saturation in the camCalib module, so that's fine. However, we don't know how to set the sharpness / hue / gamma when using the Bayer decoding.

Does anyone know how to modify these values? Modifying these value in the configuration files (https://github.com/robotology/robots-configuration/blob/master/iCubLondon01/camera/dragonfly2_config_right_bayer_640_480.ini) doesn't change anything.

Would be great if someone has an idea.
Best, Tobi

pattacini, 2018-05-21:
Hi Tobi,

I'm not very knowledgeable of this, but I can say I observed pretty much the same behavior you described: enabling Bayer results somehow in not having the full set of options available. I didn't investigate, but others may possibly chime in!

cc @barbalberto @Nicogene @fbottarel

Tobias-Fischer, 2018-05-21:
Hi Ugo, thanks for this - good to know that you don't know either :P. 
I think not having the full set of options is "by design" (somehow the hardware does not allow you to set those options if you don't use the encoding on board or something like that, I read it somewhere). But I would have thought that there is some software module (like camCalib) which allows you to post-process the images .. maybe the other guys know more :).

barbalberto, 2018-05-22:
@Tobias-Fischer if you thing the module I suggested is good for you, I'll close this QA

Tobias-Fischer, 2018-05-23:
Sure, we can close it. We actually got reasonable colors etc. after just playing with the default options available in the frameGrabberGui, but we will escape to the module in the `attention` repo in case we want to tune further.

Thanks a lot @barbalberto!

closed 2020-12-15 by pattacini