robotology/community #295: yarpdatadumper update frequency

tasbolat1, 2018-06-08:
Hello iCub community,

I have two questions:
1. What's the actual frequency of tactile data coming from any skin port of icub? (i.e /icub/skin/right_arm, etc)
2. `yarpdatadumper` is really useful software to record the data. However, I couldn't find the configuration to change the recording frequency of bottle type. Is it possible to increase the recording frequency? I know that decreasing is possible through `--downsample` parameter.

Thanks,
Tasbolat

pattacini, 2018-06-08:
Hi @tasbolat1 

Let me answer the second question.

Doing up-sampling is useless since `yarpdatadumper` simply records what is connected to. It's indeed the data source responsible for increasing the transmission frequency.

What the receiver might be doing is to virtually increase the frequency by replicating the information with data padding. However, following a good system engineering approach, this can be done profitably outside of the `yarpdatadumper` - if it is really needed - by coding a simple relay that plays as the man in the middle.

In your context, if you want to increase the frequency, you ought to better tinker with the infrastructure responsible for streaming skin data (if it's doable).

pattacini, 2018-06-08:
@towardthesea could you please report on the frequency of skin data?

tasbolat1, 2018-06-08:
Hi @pattacini,

Let me clarify: if there is a yarp program which outputs some data with `f` frequency, then the yarpdatadumper records it in `f` frequency? So, `yarpdatadumper` is not fixed to record bottle type to 50 Hz?

pattacini, 2018-06-08:
@tasbolat1 👍 yep, that's correct.

`yarpdatadumper` will record incoming data with their own original frequency.

Tobias-Fischer, 2018-06-08:
As far as I know, the frequency is 50Hz (see https://doi.org/10.1109/TAMD.2015.2507439).

tasbolat1, 2018-06-08:
Hi @Tobias-Fischer,
is it possible to increase it, at least for some parts of skin?

Tobias-Fischer, 2018-06-08:
Hi, I'd like to tell you but I really don't know .. maybe @towardthesea can help out?

closed 2020-12-15 by pattacini