robotology/community #296: Tactile sensor sensitivity

tasbolat1, 2018-06-11:
Hello, iCub community!

Right now I'm working with tactile data from its fingers. During the data gathering process, I noticed several things.
At the beginning I have the following setup:
[image]

In this experiment, the iCub fingers (middle or index) gradually touches the weighting scale. I record both weighting scale data and tactile data from the finger (the finger which in touch with the weighting scale). However, even when weighting scale shows 250-300 grams,  the tactile data stays in its baseline (no sense, hence not activated). During the process, I also use `iCubSkinGui` to visualize the tactile data and it also shows no response. 
On the other hand, side taxel of fingers (which doesn't touch the weighting scale) are more sensitive. I noticed it after just pressing the side taxels very little and visually seeing it.

In the paper https://ieeexplore.ieee.org/document/7353747/ related to latest iCub fingers, it states that the fingers' tactile  sensors can sense as small as `0.05 N`. However, it seems not like that.

Questions:
1. Are taxels in the center of the fingers  worn out in my cases (how to detect it, if so)?
2. In general, we expect the iCub tactile sensors on fingers to be in the same level of sensitivity (with some difference). I  think, Fig 9 from that paper depicts it very well. How much difference expected throughout  taxels in one finger?
3. What can I do to make it more sensitive? is there mechanical way of doing so given the time constraints?

Thanks,
Tasbolat

tasbolat1, 2018-06-13:
Anyone can answer?

lornat75, 2018-06-15:
Hi @tasbolat1. A couple of things may affect the sensitivity you measure. Firstly, make sure only one finger is touching the scale, otherwise the force is distributed (from your picture the index seems to be slightly higher so this may not be the case). Secondly, the measures from the paper have been obtained by pressing at the center of the taxel (for this purpose we mounted the fingertip so it is slightly rotated around the main axis, while the probe is vertical). In your case the scale touches an area which is probably between the taxels (check Figure 4 on the paper), therefore the sensitivity is lower.

Finally, some taxels are indeed more sensitive, we also noticed that the one at the far side are more sensitive as you experienced yourself.

I hope this helps!

closed 2020-12-08 by pattacini