robotology/community #300: :warning: [warning] YARP master not compatible with iCub master

drdanz, 2018-06-11:
YARP 3.0.0 was just released, but since we did a lot of breaking changes since the latest release, we decided to wait some more to release the iCub software, in order to ensure that everything is working properly.

Therefore at the moment robotology/icub-main and the other iCub-related `master` branch is not compatible to YARP `master` branch. If you are using the master branch, we suggest to use the tag `v2.3.72.1` in YARP.

We will close this issue as soon as this situation is resolved.

Sorry in advance if this will cause any issue.

claudiofantacci, 2018-06-11:
cc @Iaxama @towardthesea @xEnVrE

drdanz, 2018-07-31:
YARP 3.1.0 and iCub 1.12.0 were just released, icub-main master is compatible with YARP master again.
Sorry if this caused any issue.

closed 2020-12-12 by pattacini