robotology/community #301: Python bindings - missing interfaces?

emilio-cartoni, 2018-06-12:
Dear all,
I am using python bindings to control the iCub but I am having some issues:
1) How do I enable "Compliant" interaction mode?
I would expect being able to do the following, similar to what it is done in C++:

```
options = yarp.Property()
options.put("device", "remote_controlboard")
options.put("local", "/test/client/right_arm")
options.put("remote", "/"+robotName+"/right_arm")
driver = yarp.PolyDriver(options)
iint = driver.viewIInteractionMode()
iint.setInteractionMode(0, VOCAB_IM_COMPLIANT)
```

However, driver has no method viewIInteractionMode and I didn't find an alternative way to access it.
I've tried using the control interface ( driver.viewIControlMode() and  driver.viewIControlMode2()) but when I use ictrl.setImpedancePositionMode(0) in the motor gui I see that the joint goes idle in an "unknown" state.

2) Some "get" methods that rely in C++ on passing a pointer with a variable to be filled do not seem to work in Python. For example:
```
ipos.checkMotionDone(0, answer) #to check if axis 0 has completed movement
NotImplementedError: Wrong number or type of arguments for overloaded function 'IPositionControl_checkMotionDone'.
  Possible C/C++ prototypes are:
    yarp::dev::IPositionControl::checkMotionDone(int,bool *)
    yarp::dev::IPositionControl::checkMotionDone(bool *)
    yarp::dev::IPositionControl::checkMotionDone()
    yarp::dev::IPositionControl::checkMotionDone(std::vector< bool,std::allocator< bool > > &)
    yarp::dev::IPositionControl::checkMotionDone(int,std::vector< bool,std::allocator< bool > > &)
```
```
iimp.getImpedance(0,stiff,damp)
TypeError: in method 'IImpedanceControl_getImpedance', argument 3 of type 'double *'
```

3) I cannot find the constants such as "VOCAB_CM_POSITION" or "VOCAB_CM_POSITION_DIRECT", so to switch from one to the following I have to use this:
`iCtrl.setControlMode(j, yarp.Vocab_encode('pos'))`
and
`iCtrl.setControlMode(j, yarp.Vocab_encode('posd'))`

This does work, so it isn't a real issue, but I was a bit perplexed on having to resort to this instead of using something like:
`iCtrl.setControlMode(j, yarp.VOCAB_CM_POSITION)`

traversaro, 2018-06-12:
Hi @emilio-cartoni ! 

Regarding 1), I think the `viewIInteractionMode` method is missing at all in the bindings. 

See https://github.com/robotology/yarp/blob/master/bindings/yarp.i#L797 for an example on how to add `viewInterface` method .

jgvictores, 2018-06-12:
@emilio-cartoni Regarding (3) `iCtrl.setControlMode(j, yarp.VOCAB_CM_POSITION)`, it's what we recently addressed at https://github.com/robotology/yarp/pull/1696
Unfortunatley, it only works for SWIG >= 3.0.11 (careful, there are some variants of 3.0.11, so maybe even better 3.0.12 or 4 to play it safe), so you might have to include some code as mentioned [here](https://github.com/robotology/yarp/issues/1719#issuecomment-394266517) for production code.

emilio-cartoni, 2018-06-12:
Thanks, @traversaro , @jgvictores 

Ok, I have added a few lines to yarp.i to enable viewInteractionMode:
```
%include <yarp/dev/IInteractionMode.h>
+    yarp::dev::IInteractionMode *viewIInteractionMode() {
+        yarp::dev::IInteractionMode *result;
+        self->view(result);
+        return result;
+    }
```

I have also tried to solve issue (2) for one of the methods in IInteractionMode by copying similar code used for getControlMode of IControlMode.

```
%extend yarp::dev::IInteractionMode {
    yarp::dev::InteractionModeEnum getInteractionMode2(int axis) {
	yarp::dev::InteractionModeEnum mode;
	bool ok = self->getInteractionMode(axis, &mode);
        return mode;
    }
}
```
(A bit ugly as I don't know what will happen if ok is False and mode isn't filled by some values...)

As for issue (3), SWIG 3.0.08 is the latest on our Ubuntu systems, so we'll keep using Vocab.decode for a while.

Are there any plans to put IInteractionMode officially in the bindings?
Should I make a pull request with the code we will add to yarp.i?

traversaro, 2018-06-12:
>  Should I make a pull request with the code we will add to yarp.i?

Great idea, thanks!

jgvictores, 2018-06-12:
@emilio-cartoni Yes a PR (Pull Request) is the way to go! : )

BTW, you should have the three [InteractionModeEnum](http://www.yarp.it/namespaceyarp_1_1dev.html#aa6e0de626466de7cfa94283de02a8c4e) (`VOCAB_IM_COMPLIANT`, etc) for any version of SWIG, because they are defined within an `enum` and treated differently (fixed at previous PR https://github.com/robotology/yarp/pull/1655).

Nicogene, 2018-06-19:
Closed by robotology/yarp#1757

closed 2020-12-15 by pattacini