robotology/community #305: iCub face expressions doen't run

pfdominey, 2018-06-21:
Trying to run facial expression leds on the icub face
using yarpmanager to run
yarpdev pc104 --name /icub:face:raw --device serial --subdevice serialport --context faceExpressions --from serialport.ini

get the error

[ERR] (Face_Expressions) cannont run yarpdev on pc104: cannot run yarpdev on /pc104

note that other yarpdev can run on pc104, for the cameras, etc.

We changed the pc104 6 months ago or so.  Is there possibly a missing serial prot connection that we did not update?

thanks - cheers - Peter, faithful iCub_Lyon_01 user since 2009!

pfdominey, 2018-06-21:
It turns out that an environment variable was not set to include serial.ini
we changed that and now it works -
one more simple questins
what are the command line commands to change the facial expressions?

pattacini, 2018-06-21:
Hi @pfdominey 

Look at point 4 of this guide: http://wiki.icub.org/wiki/VVV08/face_expressions.
Typical values of `<cmd>` are: `hap`, `cun`, `evi`...

You'll find out more in [emotions.ini](https://github.com/robotology/icub-main/blob/master/app/faceExpressions/conf/emotions.ini) file.

closed 2020-12-15 by pattacini