robotology/community #311: how to get an image stream from the cameras?

pattacini, 2018-07-25:
@holygame commented on [Wed Jul 25 2018](https://github.com/robotology/icub-tutorials/issues/12)

I'm trying to get an image stream from the icub/icubSim cameras to use it with the ovrheadset driver .
If my understanding is correct , the ports where the images are streamed are  /icub/cam/left  and /icub/cam/right. 
i'm stuck on how to actually get the cameras to send images on those ports. For example , when controlling the head  we open the "remote_controlboard" device and connect it to the appropriate ports. What device and what interfaces do we need open to stream the output of the cameras ( for both icub and icubSim ).

Thank you for your help.

---

@pattacini commented on [Wed Jul 25 2018](https://github.com/robotology/icub-tutorials/issues/12#issuecomment-407764198)

Hi @holygame 
Welcome in robotology.

The proper place where discussing topics and asking for assistance is our Q&A repo. Therefore, I'll redirect there your request.

barbalberto, 2018-07-25:
Hi @holygame,

the idea is the same, the device you have to use is the ` remote_grabber`. You can find the documentation here: http://www.yarp.it/classyarp_1_1dev_1_1RemoteFrameGrabber.html

The parameters required are:
"remote" "/imageSourcePort"
"local" "/myPort"

Also, you can find a list of available network wrapper with `yarpdev --list`

holygame, 2018-07-26:
hi @barbalberto , thanks to you I was finally able to stream the images and see them through OpenCV!
Now I'm trying to stream them to the oculus headset.
the problem i'm facing is how to open the ovrheadset driver , there are 6 parameters that are required ;
- "tfLocal"
- "tfRemote"
- "tf_left_hand_frame"
- "tf_right_hand_frame"
- "tf_root_frame"
- "stick_as_axis"
- "gui_elements"

i couldn't find any documentation about those , could you please give me some hints ? 
thank you so much for the help!

barbalberto, 2018-07-27:
I ping @aerydna who made that device to get more info about it (and didn't write the documentation!! :imp:  )

aerydna, 2018-07-27:
documentation added via pr. 

* Parameters accepted in the config argument of the open method:
* |   Parameter name      | Type   | Units | Default Value | Required  | Description                               | Notes |
* |:---------------------:|:------:|:-----:|:-------------:|:---------:|:-----------------------------------------:|:-----:|
* | tfLocal               | string |       |               | yes       | local port name receiving and posting tf  |       |
* | tfRemote              | string |       |               | yes       | name of the transformServer port          |       |
* | tf_left_hand_frame    | string |       |               | Yes       | name of the left hand frame               |       |
* | tf_right_hand_frame   | string |       |               | yes       | name of the right hand frame              |       |
* | tf_root_frame         | string |       |               | yes       | name of the root frame                    |       |
* | stick_as_axis         | bool   |       |               | yes       | if axes shoud be published as sticks      |       |
* | gui_elements          | int    |       |               | yes       | number of the gui element to visualize    |       |

Gui Groups parameters
* |   Parameter name      | Type   | Units | Default Value | Required  | Description               | Notes |
* | width                 | double | pixel |               | yes       | width of the widget       |       |
* | height                | double | pixel |               | yes       | height of the widget      |       |
* | x                     | double | pixel |               | yes       | x position of the widget  |       |
* | y                     | double | pixel |               | yes       | y position of the widget  |       |
* | z                     | double | pixel |               | yes       | z position of the widget  |       |
* | alpha                 | double |       |               | yes       | alpha value of the widget |       |

holygame, 2018-08-06:
I've tried different value for the parameters but I can't get the device to work.  I'm just interested in connecting the cameras  of the icub with the oculus  , what values of this parameters I should use ?

barbalberto, 2018-08-06:
Please provide a description of the error you get when launching the device.

holygame, 2018-08-06:
Could you please check if this default values make sense ? in the meantime i'll prepare the screenshot of the error 

 
```
* |   Parameter name      | Type   | Units | Default Value | Required  | Description                               | Notes |  
* | tfLocal               | string |       |      local_oculus        | yes       | local port name receiving and posting tf  * | tfRemote              | string |       |      ?? oculus ??        | yes       | name of the transformServer port          
* | tf_left_hand_frame    | string |       |     left_hand      | Yes       | name of the left hand frame                
* | tf_right_hand_frame   | string |       |      right_hand  | yes       | name of the right hand frame              |       | 
* | tf_root_frame         | string |       |       torso   | yes   | name of the root frame                    |       | 
* | stick_as_axis         | bool   |       |      0         | yes       | if axes shoud be published as sticks      |       | 
* | gui_elements          | int    |       |      3         | yes       | number of the gui element to visualize    |       | 
```

holygame, 2018-08-06:
```
||| clearing context
||| adding context [myModule/conf]
||| configuring
||| default config file specified as myModule.ini
||| checking [C:\Users\AnDy\Yarp_Projects\ICubCamToOculus\build\Release\myModule.ini] (pwd)
||| found C:\Users\AnDy\Yarp_Projects\ICubCamToOculus\build\Release\myModule.ini
yarp: Port / active at tcp://10.0.0.68:10055/
||| finding file [icubEyes[.ini]
IN|||F checking [C:\Users\AnDy\Yarp_Projects\ICubCamToOculus\build\Release\icubEyes.ini] (pwd)
O||| found C:]\Users\AnDy\Yarp_Projects\ICubCamToOculus\build\Release\icubEyes.ini
Listening to terminal (type "quit" to stop module).
[DEBUG]config is  (device remote_grabber) (local "/oculus/display/right:i") (remote "/icub_sim/Cam/right")
yarp: Port /oculus/display/right:i active at tcp://10.0.0.68:10057/
[INFO]connecting  /oculus/display/right:i  to  /icub_sim/Cam/right
[ERROR]cannot connect  /oculus/display/right:i  to  /icub_sim/Cam/right
[INFO]created device <remote_grabber>. See C++ class yarp::dev::RemoteFrameGrabber for documentation.
[ERROR]unable to dynamic cast device to IFrameTransform interface
yarpdev: ***ERROR*** driver <ovrheadset> was found but could not open
ovrheadset not ready
[INFO]RFModule failed to open.
```

here is the configuration file used 

```
device ovrheadset
tfDevice remote_grabber
tfLocal /oculus/display/right:i
tfRemote /icub_sim/Cam/right
tf_left_hand_frame  /icub_sim/arm/left
tf_right_hand_frame /icub_sim/arm/right
tf_root_frame str
stick_as_axis false
gui_elements 0
```

barbalberto, 2018-08-06:
The errror says
`[ERROR]cannot connect  /oculus/display/right:i  to  /icub_sim/Cam/right`
so either the port producing the image is not open yet or its name is different.

Check the port name with `yarp name list`

holygame, 2018-08-06:
I've tried with the correct port name but still have problems , it's also weird that i have to use one port for the images but i have 2 port streaming the images ( left and right eyes )

```
||| clearing context
||| adding context [myModule/conf]
||| configuring
||| default config file specified as myModule.ini
||| checking [C:\Users\AnDy\Yarp_Projects\ICubCamToOculus\build\Release\myModule.ini] (pwd)
||| found C:\Users\AnDy\Yarp_Projects\ICubCamToOculus\build\Release\myModule.ini
yarp: Port / active at tcp://10.0.0.68:10055/
||| finding file [icubEyes.ini]
[||| checking [CINFO:\Users\AnDy\Yarp_Projects\I]CubCaLimToOculus\build\Reslease\icubEyes.init] (pwd)
en||| found C:\Users\AnDy\Yiarp_Projects\ICubCamToOculus\build\Release\icubEyes.inin
g to terminal (type "quit" to stop module).
[DEBUG]config is  (device remote_grabber) (local "/oculus/display/right:i") (remote "/icubSim/cam/right")
yarp: Port /oculus/display/right:i active at tcp://10.0.0.68:10057/
[INFO]connecting  /oculus/display/right:i  to  /icubSim/cam/right
yarp: Receiving input from /icubSim/cam/right to /oculus/display/right:i using tcp
yarp: Sending output from /oculus/display/right:i to /icubSim/cam/right using tcp
[INFO]created device <remote_grabber>. See C++ class yarp::dev::RemoteFrameGrabber for documentation.
[ERROR]unable to dynamic cast device to IFrameTransform interface
yarpdev: ***ERROR*** driver <ovrheadset> was found but could not open
yarp: Removing input from /icubSim/cam/right to /oculus/display/right:i
yarp: Removing output from /oculus/display/right:i to /icubSim/cam/right
ovrheadset not ready
[INFO]RFModule failed to open.`

```
config

```
device ovrheadset
tfDevice remote_grabber
tfLocal /oculus/display/right:i
tfRemote /icubSim/cam/right
tf_left_hand_frame  /icub_sim/arm/left
tf_right_hand_frame /icub_sim/arm/right
tf_root_frame str
stick_as_axis false
gui_elements 0
```

barbalberto, 2018-08-07:
The config file is not correct
`tfDevice remote_grabber`     tfDevice has to be the device handling the Transform Frames, while the `remote_grabber` is for the images.
@aerydna can you assist @holygame in creating a correct config file?

holygame, 2018-08-08:
thank you @aerydna and @barbalberto it's working perfectly !

closed 2020-12-08 by pattacini