robotology/community #312: cartesian controller, help for attachTipFrame()

valeriosperati, 2018-07-26:
Please, about cartesian controller (with icub simulator) I was trying the code here [Defining a Different Effector](http://www.icub.org/software_documentation/icub_cartesian_interface.html#sec_cart_tipframe).

Question 1: which headers should I include for class iCubFinger?  
I tried them but I'm not sure:
`#include <iCub/iKin/iKinFwd.h>`
`#include <iCub/iKin/iKinHlp.h>`
`#include <iCub/iKin/iKinInv.h>`
`#include <iCub/iKin/iKinIpOpt.h>`
`#include <iCub/iKin/iKinSlv.h>`
`#include <iCub/iKin/iKinVocabs.h>`

Question 2: should I include some particular library in CMakeLists?
Right now, when I try to compile I get several undefined reference:
 
```
CMakeFiles/exe.dir/myIcub.cpp.o: In function `myIcub::initParam(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':         
myIcub.cpp:(.text+0x5678): undefined reference to `iCub::iKin::iCubFinger::iCubFinger(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
myIcub.cpp:(.text+0x763b): undefined reference to `vtable for iCub::iKin::iCubFinger'
myIcub.cpp:(.text+0x76a8): undefined reference to `iCub::iKin::iKinLimb::~iKinLimb()'
CMakeFiles/exe.dir/myIcub.cpp.o: In function `iCub::iKin::iCubFinger::~iCubFinger()':
myIcub.cpp:(.text._ZN4iCub4iKin10iCubFingerD2Ev[_ZN4iCub4iKin10iCubFingerD5Ev]+0x3): undefined reference to `vtable for iCub::iKin::iCubFinger'
myIcub.cpp:(.text._ZN4iCub4iKin10iCubFingerD2Ev[_ZN4iCub4iKin10iCubFingerD5Ev]+0x5f): undefined reference to `iCub::iKin::iKinLimb::~iKinLimb()'
CMakeFiles/exe.dir/myIcub.cpp.o: In function `iCub::iKin::iCubFinger::~iCubFinger()':
myIcub.cpp:(.text._ZN4iCub4iKin10iCubFingerD0Ev[_ZN4iCub4iKin10iCubFingerD5Ev]+0x3): undefined reference to `vtable for iCub::iKin::iCubFinger'
myIcub.cpp:(.text._ZN4iCub4iKin10iCubFingerD0Ev[_ZN4iCub4iKin10iCubFingerD5Ev]+0x5e): undefined reference to `iCub::iKin::iKinLimb::~iKinLimb()'
collect2: error: ld returned 1 exit status
CMakeFiles/exe.dir/build.make:830: recipe for target 'exe' failed
```

I guess I need the proper library in TARGET_LINK_LIBRARIES, which currently is:
`TARGET_LINK_LIBRARIES(exe ${OpenCV_LIBS} ${YARP_LIBRARIES} itpp Qt5::Widgets)`

Thanks, Valerio

pattacini, 2018-07-26:
Hi @valeriosperati 

You ought to:
1. include `iKinFwd.h`
2. link against `ctrlLib`, thus:
    ```cmake
    target_link_libraries(exe ctrlLib)
    ```

pattacini, 2018-07-26:
link against `iKin`, sorry for the typo

valeriosperati, 2018-07-26:
Perfect! Thanks

valeriosperati, 2018-07-27:
Please, I need clarification about the use of code, in particular I'm comparing [Defining a Different Effector](http://www.icub.org/software_documentation/icub_cartesian_interface.html#sec_cart_tipframe) and [How to use the iCubFinger forward kinematics](https://github.com/robotology/QA/issues/152). 

It is not clear to me if  the following piece of code, computing the Matrix  tipFrame
```
iencs->getEncoders(encs.data());
yarp::sig::Vector joints;
finger.getChainJoints(encs,joints);
yarp::sig::Matrix tipFrame=finger.getH((M_PI/180.0)*joints);

Vector tip_x=tipFrame.getCol(3);
Vector tip_o=yarp::math::dcm2axis(tipFrame);
icart->attachTipFrame(tip_x,tip_o); 
```
must be used only once, during initialisation of cartesian controller, or if I need to recompute it everytime before a call to 
`icart->goToPose(xd,od);`

As I read [here](https://github.com/robotology/QA/issues/152),  the helper method getChainJoints() is based on the values of encoders, so I was wondering if I need to refresh it -through a call to iencs->getEncoders(encs.data())- before each new cartesian command.
Thanks

pattacini, 2018-07-27:
Anytime you move the finger's joints, then you have to update the matrix `tipFrame` and let the Cartesian Controller know about that.

In practice, this won't happen so often, thereby the workflow could be:
1. Put the finger in the configuration whose tip you'd like to control. 
1. Read the finger's joints once (making sure you'll obtain a fresh update).
1. Compute the matrix `tipFrame`.
1. Call `attachTipFrame` only once.
1. Run your control by calling `goToPose` as many times as you need, in the assumption that the finger won't move.
1. Remember to restore the default control point by means of `removeTipFrame` (or through context switching).

valeriosperati, 2018-07-27:
Ok, so if I have to recompute for some reason the index finger joints, I have to redo all the steps you say, right?.
A last question: in this last case do I have to call removeTipFrame() before calling again AttachTipFrame()?
Thanks

pattacini, 2018-07-27:
1. Yep 👍 
1. Nope, you can skip the `removeTipFrame` and proceed straight to `attachTipFrame`.

valeriosperati, 2018-07-30:
Thanks!
Valerio

closed 2020-12-08 by pattacini