robotology/community #313: ⚠️ [warning] Releases in progress (YARP and iCub)

drdanz, 2018-07-31:
We are about to release YARP and iCub, please do not update your robot or push new commits until new instructions.

PeterBowman, 2018-07-31:
(reporting here for awareness and urgency, excuse me if I'm wrong) @drdanz I found out that [YARP_idl_tools](https://github.com/robotology/yarp/blob/62d24814a32a7fb4f30c8c0eeaf3247111178853/src/idls/CMakeLists.txt#L14) inserts a local path (i.e. relative to the build directory) in the **installed** `YARP_idl_toolsConfig.cmake` file (`/usr/local/lib/YARP_idl_tools`). **Affects YARP v3.0.1**. See [this Travis build](https://travis-ci.org/roboticslab-uc3m/openrave-yarp-plugins/jobs/410093683#L1155), also confirmed on my PC.

drdanz, 2018-07-31:
Fixed, in time for the release! thanks @PeterBowman

PeterBowman, 2018-07-31:
Glad to help, @drdanz :). A few more words regarding https://github.com/robotology/yarp/commit/479d0755d64443c869454c295c9b1cdd311f031a:

* `YARP_MODULE_DIR` is generated by the root package `YARP`, so I guess it won't work if I do `find_package(YARP_idl_tools)` (but I'm not a user of this component, perhaps it's not even intended to work this way).

* Could you backport this commit to YARP 3.0.x (v3.0.2)? Any build that consumes YARP v3.0.1 will fail because of that bug.

traversaro, 2018-07-31:
Related note for the [`robotology-superbuild`](https://github.com/robotology/robotology-superbuild) users:  due to che breaking changes in YARP 3, it is possible that a few repository will need to be released after the release of `YARP` and the matching release of `icub-main`: when everything is supposed to work well, I will update this issue as well.

drdanz, 2018-07-31:
> YARP_MODULE_DIR is generated by the root package YARP, so I guess it won't work if I do find_package(YARP_idl_tools) (but I'm not a user of this component, perhaps it's not even intended to work this way).

That's not how it is supposed to be used, anyway I don't thin

drdanz, 2018-07-31:
> Could you backport this commit to YARP 3.0.x (v3.0.2)?

Sorry, devel was already merged in master, and there won't be a 3.0.2 release...

drdanz, 2018-08-02:
To be honest i don't like the master-devel branch layout... I believe that "devel" should be "master" and there should be branches for the stable releases. Unfortunately current layout does not allow supporting more than one "stable" release. Anyway I pushed a "yarp-3.0" branch including the commit, I don't know if we will ever release a YARP 3.0.2, but you can use that branch instead of "master" if you want.

PeterBowman, 2018-08-02:
That's great, @drdanz, thank you! I'm totally fine with pointing at that hotfix commit (https://github.com/robotology/yarp/commit/30dce540494b8cda5d9d40018384c047a8b47fe6) in our scripts and install guides, therefore you may want to delete the new branch (to not interfere too much with the current layout and to keep the list of branches clean for now).

drdanz, 2018-08-02:
@PeterBowman cool, I think I'll keep the branch for the moment, so if you see any commits in master or devel that you think it should be in yarp-3.0 branch let me know.

closed 2020-12-12 by pattacini