robotology/community #316: Upcoming new dependencies for iDynTree

traversaro, 2018-08-02:
In the next days, a [new shiny URDF parser](https://github.com/robotology/idyntree/pull/460) with [much better diagnostic errors](https://github.com/robotology/idyntree/pull/460#issuecomment-400845404) will land on iDynTree `devel` branch, contributed by @francesco-romano at [DeepMind](https://deepmind.com/). 

The new parser is  based on the MIT-licensed [libxml2 library](http://xmlsoft.org/): as  a consequence, libxml2 will become a new **required dependency** of iDynTree, so if you are using the *devel* branch of iDynTree, you will need to have `libxml2` installed in your system. Check the following for the detailed in structions on the reccomended way of installing it. 

# Debian/Ubuntu 
~~~
sudo apt install libxml2 
~~~

# macOS Homebrew 
~~~
brew install libxml2
~~~

# Windows 

If you are using `vcpkg` to manage your dependencies, you can install libxml2 via `vcpkg` : 
~~~
./vcpkg install --triplet x64-windows libxml2 
~~~
where `x64-windows` should be changed to `x86-windows` if you are targeting 32-bit binaries. 

If you tipically install your dependencies using the [YARP and ICUB dependencies installers](http://wiki.icub.org/wiki/Downloads), we recommend you to use the new [robotology-additional-dependencies](https://github.com/robotology-playground/robotology-additional-dependencies/releases/tag/v0.1.0), that is a new installer of dependencies generated by vcpkg that contains `libxml2`. This installer is automatically accounting for enviroment variables, similary to how the YARP and ICUB dependencies installers do. 

Please report any issue or problem regarding to this, thanks!

francesco-romano, 2018-08-02:
@traversaro note that macOS already ships with `libxml2`. 
If brew is used, it should be specified in some way to Make, otherwise I think the default system one get picked.

traversaro, 2018-08-02:
@francesco-romano so do you think it make sense **not** to install libxml2 using homebrew to avoid confusion?

francesco-romano, 2018-08-02:
It is a good question.

From my investigations I found the following: (macOS 10.13.6):
- version: 2.9.4
- library is in /usr/lib/libxml2
- headers in /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk/usr/include/libxml2 but it might be that I did not symlinked the standard libraries in /usr (it is something you do with Xcode. Anyway, this implies that you need Xcode, but I think this requirement should be verified in all cases).

The question is: what version of libxml2 is shipped in previous OS versions?
I do not know the frequency of updates of the libraries in macOS -  note that 2.9.4 is 2 years old (https://gitlab.gnome.org/GNOME/libxml2/tags)

So we might want to setup some instructions to use brew, just to stay on the safe side.

traversaro, 2018-08-03:
I checked and `libxml2` is present in most recent versions of Developer SDKs on macOS . I assume CMake has no trouble in finding it, right? For this reason, I updated the instructions relatively to `macOS` . 

By the way, the parser have been merged in the `devel` branch, see https://github.com/robotology/idyntree/pull/460 .

traversaro, 2018-08-03:
The parser will be part of the upcoming iDynTree 0.11 .

francesco-romano, 2018-08-03:
Sounds good to me. Let's keep it in this way. If in the future we realise we need a more recent version, we can switch to the one provided by homebrew

closed 2020-12-12 by pattacini