robotology/community #317: Using GazeEvent callback in Python

pattacini, 2018-08-02:
@ddetommaso commented on [Thu Aug 02 2018](https://github.com/robotology/yarp/issues/1824)

**Describe the bug**
I am trying to use the event callback for the IGazeControl class but without success. Whenever I call the method registerEvent I got the following error:

TypeError: in method 'IGazeControl_registerEvent', argument 2 of type 'yarp::dev::GazeEvent &'

**To Reproduce**

> import yarp
> 
> class GazeMotionDone(yarp.GazeEvent):
> 
>     def __init__(self):
>         pass
> 
>     def gazeEventCallback(self):
>         pass
> 
> class GazeController:
> 
>     def __init__(self):
>         self.iGaze = None
>         props = yarp.Property()
>         props.put("device","gazecontrollerclient")
>         props.put("local","/gaze_client")
>         props.put("remote","/iKinGazeCtrl")
>         self.driver = yarp.PolyDriver()
>         self.driver.open(props)
>         self.iGaze = self.driver.viewIGazeControl()
>         a = GazeMotionDone()
>         self.iGaze.registerEvent(a)

**Expected behavior**
I would expect that GazeMotionDone class is derived from yarp.GazeEvent

**Configuration (please complete the following information):**
 - OS: Ubuntu 16.04
 - yarp version: 2.3.72
-  compiler: gcc 5.4.0

Please let me know if I am missing something.

pattacini, 2018-08-03:
Hi @ddetommaso 

I'm not a Python user, hence cannot be of help here.
One possibility could be that `swig` didn't export correctly all that's needed.

Any Python/swig expert willing to chime in?

pattacini, 2018-09-17:
🔔 Gentle ping to all developers out there eager to help 😉

traversaro, 2018-09-17:
I know that this kind of behaviors are typically implemented using the [`director`'s SWIG feature](http://www.swig.org/Doc3.0/Python.html#Python_nn34). I guess we should add a `%feature("director") yarp::dev::GazeEvent` somewhere after the actual inclusion of `IGazeControl.h` in https://github.com/robotology/yarp/blob/b7983c941952a3aafa7de7dd4aeeb574c1865799/bindings/yarp.i#L426 .

closed 2020-12-08 by pattacini