robotology/community #322: OpenCV and how to convert Image to Mat

kt10aan, 2018-09-07:
In some older code there was: (I am leaving out the if port open checks)

```
1 bool YarpImageProvider::get(cv:Mat& frame) {
2       yarp::sig::ImageOf<yarp::sig::PixelRgb> *image = imagePort.read(true);
3        if (image) {
4           cv::Mat rgbframe((IplImage*)image->getIplImage;
5           cv::cvtColor(rgbframe,frame, CV_RGB2BGR);
```

and was working fine until the last upgrade started a chain of dependencies update and now we are using OpenCV3. The above code was written for OpenCV 2.4.9, so quite ancient and doesn't accept any Ipl -based methods.

What I did is replacing line 4 with:
```
cv::Mat rgbframe(image->height, image->width, 21, image);
```

However, when running the software is segfaults. Type 21 is ``CV_32FC3``. I tried with other image types but most of them give me the same segfault and a few other give me a garbled image.

My question is: what would be the correct line 4 to convert the YARP image to a ``cv::Mat``?

claudiofantacci, 2018-09-07:
Hi @kt10aan, I doesn't change anything from OpenCV 2.* to version 3.*.

With the following two lines of code you can wrap a `yarp::sig::ImageOf<yarp::sig::PixelRgb>` with a `cv::Mat`, i.e. they share the same memory representing pixel values.

```cpp
yarp::sig::ImageOf<yarp::sig::PixelRgb> *yarp_img = imagePort.read(true);
cv::Mat img = cv::cvarrToMat(static_cast<IplImage*>(yarp_img.getIplImage()));
```

kt10aan, 2018-09-07:
@claudiofantacci , many thanks! it worked! (but it is ''yarp_img->getIplImage'').
Again, many thanks:)

pattacini, 2018-09-09:
See also #122.

closed 2020-12-08 by pattacini