robotology/community #325: c++: print yarp version

valeriosperati, 2018-10-18:
Please,
is it possible in C++ to print the current yarp version?

For OpenCV I use CV_VERSION
For QT I use QT_VERSION_STR

Is there some variable for YARP?
Thanks

PeterBowman, 2018-10-18:
Just include the `<yarp/conf/version.h>` header (configured by CMake) ([ref](https://github.com/robotology/yarp/blob/d1a2effb61bd8888e7b19701d2b6bd24e4f30dd4/src/libYARP_conf/include/yarp/conf/version.h.in)). You'll probably want `YARP_VERSION_SHORT`.

valeriosperati, 2018-10-18:
Perfect!
Thanks.

drdanz, 2018-10-18:
Please note that `YARP_VERSION_SHORT` is the version that you use to build your program.
There is not, at the moment, a way to know the version of the library *loaded* when you start the program, depending on the setup, it might be different. In most cases it should be the same though...

closed 2020-12-08 by pattacini