robotology/community #329: Robot goes in hardware fault when using VOCAB_IM_COMPLIANT motorControlImpedance

caicaixia, 2019-01-08:
Dear Sir,

Today I tried the tutorial demo motorControlImpedance for the impedance control of iCub.

Its is very simple demo:
```
if (times%2)
        {
             // set the elbow joint only in compliant mode
             ictrl->setPositionMode(3);
             iint->setInteractionMode(3,VOCAB_IM_COMPLIANT);
             // set new reference positions
             command[0]=-50;
             command[1]=20;
             command[2]=-10;
             command[3]=60;
        }
        else
        {
             // set the elbow joint in stiff mode
            ictrl->setPositionMode(3);
             iint->setInteractionMode(3,VOCAB_IM_STIFF);
             // set new reference positions
             command[0]=-20;
             command[1]=40;
             command[2]=-10;
             command[3]=30;  
        }
```

However I have few problems:
1: After icub went to the `VOCAB_IM_COMPLIANT`, then the elbow joint will show a hardware fault. Then even it changes back to `VOCAB_IM_STIFF` mode, its seems like joint3 is not able to move again.

[image]

2: What is the mean of new reference positions? Joint3 will go to the new reference position or not? From the demo, I saw the command[3]=60; or command[3]=30; are never reached, it seems like the elbow joint is not controlled. This is maybe due to the problem 1.

So how can I go back to position control stiff mode after I did compliant mode.  I mean how let the icub assume the normal position control?

julijenv, 2019-01-08:
Hi @caicaixia ,
my colleague will return to you ASAP, I guess the thread will be moved to another repository cox it is not related to any hardware issue.
Just as a matter of clarity, are you using the overall demo file or just a part of it?!?!? ( I mean https://github.com/robotology/icub-tutorials/blob/master/src/motorControlImpedance/tutorial_arm_joint_impedance.cpp)

pattacini, 2019-01-08:
Hi @caicaixia 

As per our contributing guidelines of [icub-support][1], I moved here in [QA][2] your issue.
We are going to triage it soon, stay tuned.

(please, in the future avoid assigning issues directly to specific people since traging is a mantainers' task)

[1]: https://github.com/robotology/icub-support/blob/master/.github/SUPPORT.md
[2]: https://github.com/robotology/QA/issues

caicaixia, 2019-01-08:
Hi @julijenv,  I just downloaded this demo and use the whole demo.

caicaixia, 2019-01-08:
Hi @pattacini  Sorry for ur trouble.  I WILL remember in future.

pattacini, 2019-01-08:
@caicaixia it might be that you did not launch `wholeBodyDynamics` (`WBD` in short) prior to running the demo. `WBD` is required to deal with the F/T computations. For safety reasons, if `WBD` is unavailable, the joints will turn red within the GUI signaling the problem. To get them back functional, you have first to put them in `idle` (yellow color) and then select `run`.

You can find more information on this visiting http://wiki.icub.org/wiki/Force_Control, although the page is not really up-to-date with our most recent software. For this, @randaz81 might chime in giving further hints and heads-up.

caicaixia, 2019-01-09:
Hello @pattacini , 

1:   I tried the run `wholeBodyDynamics`, but it has the same error. The join went to idle hardware fault.
I run the nodes as:
```
 <name>Tutorial on Impedance Control iCub</name>
  <dependencies>
      <port>/icub/yarprobotinterface</port>
  </dependencies>
  <module>
      <name>yarplogger</name>
      <parameters>--start --no_stop</parameters>
      <node>icubsrv</node>
  </module>
  <module>
	<name>wholeBodyDynamics</name>
	<parameters>--no_legs</parameters>
	<node>icubsrv</node>
	<stdio>icubsrv</stdio>
	<tag>wholeBodyDynamicsTag</tag>
  </module>
  <module>
	<name>gravityCompensator</name>
	<parameters>--no_legs</parameters>
	<node>icubsrv</node>
	<stdio>icubsrv</stdio>
	<tag>gravityCompensatorTag</tag>
  </module>
  <module>
      <name>motorControlImpedance</name>
      <parameters>--robot icub</parameters>
      <environment>YARP_FORWARD_LOG_ENABLE=1</environment>
      <node>icubsrv</node>
  </module>
```

Here is my yarpmanger:
[image]

2:  In code,  I also tried to  send joints in IDLE/RUN (the same as clicked the button), but it is also not working :(   I don't know what can I do anymore.
```
// set the elbow joint in stiff mode
//             ictrl->setControlMode(3, VOCAB_CM_FORCE_IDLE); // click the Idle button
             ictrl->setControlMode(3, VOCAB_CM_POSITION);
             iint->setInteractionMode(3,VOCAB_IM_STIFF);
```

3:  My purpose is to do a human comliant interaction with iCub,  after the contact, the icub can go back to his work again (like go to desire position).
Similar like this video:  https://www.youtube.com/watch?v=RhtVOyKuL_0
https://www.youtube.com/watch?v=LMGSok5tN4A

traversaro, 2019-01-09:
Hi @caicaixia, I suspect you are missing a few connections between the robot and the `wholeBodyDynamics` module. I suggest you to remove manual connections involving any port with the `/wholeBodyDynamics` prefix and just pass the `--autoconnect` option to both `wholeBodyDynamics` and `gravityCompensator` module. 
In particular, I suggest to just use the following application template: https://github.com/robotology/icub-main/blob/4f5881acafab693f763dd277f87a8996fdea01a3/app/iCubStartup/scripts/iCubStartup.xml.template#L68 , eventually disabling `iKinGazeCtrl` and `iKinCartesianSolver` if you are not using them. 
If you have further problems, please also report the output of the `wholeBodyDynamics` and `gravityCompensator` modules that you should see in the `yarplogger`, thanks!

caicaixia, 2019-01-09:
hello @traversaro @pattacini,  
Thanks for the help.  The demo seems working now.  It doesn't go to the HARDWARE_FAULT anymore after the compliant mode :)

pattacini, 2019-01-09:
Glad to hear that!

closed 2020-12-08 by pattacini