robotology/community #330: fingersTuner operation

dhgarcia, 2019-01-10:
We have a question for the operation of fingersTuner.

We try to run from yarpmanager the modules for iCubStartup  (yarprobotinterface, iKinCartesianSolver, iKinGazeCtrl, wholeBodyDynamics, gravityCompensator, fingersTuner, imuFilter)

fingersTuner always fail.
from terminal it fails with the following message:
[ERROR]group [left_hand] is missing!
[INFO]RFModule failed to open.

The hand does work with yarprobotinterface and yarpmotorcontrolgui,
How can we debug the issue?

Thanks and best regards

pattacini, 2019-01-10:
Hi @dhgarcia 

Most likely, `fingersTuner` starts off without loading the correct configuration file, which instead seems to lack the `[left_hand]` group, as reported.

A correct configuration file can be found in the [`icub-main repo`][1].

To check which configuration file will be loaded when launched you can do (see #42 for more details):
```sh
$ yarp resource --context fingersTuner --from config.ini
```

[1]: https://github.com/robotology/icub-main/blob/master/app/fingersTuner/conf/config.ini

dhgarcia, 2019-01-11:
Thank you @pattacini 
The configuration file loaded was missing both hands.

We updated it with the one in the `icub-main` repo and everything works fine now.

Thanks

closed 2020-12-15 by pattacini