robotology/community #331: Robot error when run YARPROBOTINTERFACE

caicaixia, 2019-01-16:
Hello guys,

Today we are doing experiments with iCub.  At beginning everything is ok.  Then we restart iCub after sometime, it can not run anymore.

When run `yarprobotinterface`, it shows many errors, and the robot can not connected anymore.
[image]

[image]

what can we do now?

caicaixia, 2019-01-16:
@tasbolat1

julijenv, 2019-01-16:
Hi @caicaixia and @tasbolat1 , could you please try to join us the entire log file of the failure to get more inside of from where come from the failure, thx in advance

caicaixia, 2019-01-16:
Hello @julijenv , how to get the entire logfile? from the yarplogger?

julijenv, 2019-01-16:
yes you should have a save log file in one of the tab.

caicaixia, 2019-01-16:
[yarprunlog_16_01_2019_19_57_01.log](https://github.com/robotology/QA/files/2764041/yarprunlog_16_01_2019_19_57_01.log)

caicaixia, 2019-01-16:
Hello @julijenv , any idea about it?   It suddenly becomes like this.  We didn't change or update anything

julijenv, 2019-01-16:
sorry lunch break I will get right back to you soon

tasbolat1, 2019-01-16:
Hi @julijenv, any news about the log?

julijenv, 2019-01-16:
seems like the right leg cant see 2 2FOC boards, could you check the connections or just remove this part to see if the robot strat correctly without the right leg?!?!

> #STRING_START#>embObjMC BOARD  right_leg-eb9-j4_5 Missing OTHER_CONTROL_PARAMETERS.DeadZone parameter. I'll use default value. (see documentation for more datails) 
<#STRING_END#>
675
3,141126
4
<#STRING_START#>Verbose parameter not found for board right_leg-eb9-j4_5in bottle"2FOC" (HasHallSensor 1 1) (HasTempSensor 0 0) (HasRotorEncoder 1 1) (HasRotorEncoderIndex 0 0) (HasSpeedEncoder 0 0) (RotorIndexOffset 0 0) (MotorPoles 8 8) 
<#STRING_END#>
676
3,141154
4
<#STRING_START#>In  right_leg-eb9-j4_5  there isn't 2FOC.Verbose filed. For default it is enabled 
<#STRING_END#>
677
3,141177
4
<#STRING_START#>AutoCalibration parameter not found for board right_leg-eb9-j4_5in bottle"2FOC" (HasHallSensor 1 1) (HasTempSensor 0 0) (HasRotorEncoder 1 1) (HasRotorEncoderIndex 0 0) (HasSpeedEncoder 0 0) (RotorIndexOffset 0 0) (MotorPoles 8 8) 
<#STRING_END#>
678
3,141206
4
<#STRING_START#>In  right_leg-eb9-j4_5  there isn't 2FOC.AutoCalibration filed. For default it is disabled 
<#STRING_END#>
679
3,141229
2
<#STRING_START#>EthResource::verifyBoardPresence() found BOARD right_leg-eb9-j4_5 with IP 10.0.1.9 after 0.00154901 seconds 
<#STRING_END#>
680
3,141253
2
<#STRING_START#>EthResource::verifyBoardTransceiver() has validated the transceiver of BOARD right_leg-eb9-j4_5 with IP 10.0.1.9 
<#STRING_END#>
681
3,141294
2
<#STRING_START#>EthResource::cleanBoardBehaviour() has cleaned the application in BOARD right_leg-eb9-j4_5 with IP 10.0.1.9 : config mode + cleared all its regulars 
<#STRING_END#>
682
3,151067
2
<#STRING_START#>EthResource::setTimingOfRunningCycle() for BOARD right_leg-eb9-j4_5 with IP 10.0.1.9 has succesfully set: cycletime = 1000 usec, RX DO TX = ( 400 300 300 ) usec and TX rate = 5  every cycle 
<#STRING_END#>
683
3,161358
3
<#STRING_START#>EthResource::askBoardVersion() found BOARD right_leg-eb9-j4_5 @ IP 10.0.1.9 of type ems4 with FW = ver 3.20 built on 2018 Mar 19 10:30 
<#STRING_END#>
684
3,192193
2
<#STRING_START#> from BOARD 10.0.1.9 (right_leg-eb9-j4_5), src LOCAL, adr 0, time 37s 405m 373u: (code 0x05000015, par16 0x2000 par64 0x0000000000ff0000) -> CFG: EOtheEncoderReader can be correctly configured. p16&0xf000: number of joint; primary encs: failure mask in p16&0x000f and errorcodes in p64&0x0000ffff; secondary encs: failure mask in p16&0x00f0 and errorcodes in p64&0xffff0000 + .
<#STRING_END#>
685
3,192303
3
<#STRING_START#> from BOARD 10.0.1.9 (right_leg-eb9-j4_5) @ 37s 405m 524u: CAN discovery has started for 2 eobrd_foc boards on (can1map, can2map) = (0x0006, 0x0000) with target can protocol ver 1.6 and application ver 0.0.0.
<#STRING_END#>
686
3,663648
5
<#STRING_START#>theNVmanager::Impl::wait() had a timeout for BOARD right_leg-eb9-j4_5 IP 10.0.1.9 and nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult 
<#STRING_END#>
687
3,663754
5
<#STRING_START#>theNVmanager::Impl::command() fails a wait() from IP 10.0.1.9 for nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult 
<#STRING_END#>
688
3,694188
2
<#STRING_START#> from BOARD 10.0.1.9 (right_leg-eb9-j4_5), src LOCAL, adr 0, time 37s 905m 390u: (code 0x05000015, par16 0x2000 par64 0x0000000000ff0000) -> CFG: EOtheEncoderReader can be correctly configured. p16&0xf000: number of joint; primary encs: failure mask in p16&0x000f and errorcodes in p64&0x0000ffff; secondary encs: failure mask in p16&0x00f0 and errorcodes in p64&0xffff0000 + .
<#STRING_END#>
689
3,694292
5
<#STRING_START#> from BOARD 10.0.1.9 (right_leg-eb9-j4_5) @ 37s 905m 548u: CAN discovery after 500 ms has detected 2 missing eobrd_foc boards in CAN1:
<#STRING_END#>
690
3,694330
5
<#STRING_START#>1 of 2: missing eobrd_foc BOARD 10.0.1.9:CAN1:1
<#STRING_END#>
691
3,694357
5
<#STRING_START#>2 of 2: missing eobrd_foc BOARD 10.0.1.9:CAN1:2
<#STRING_END#>

julijenv, 2019-01-16:
please try to contact me on skype: my id is icub.support

caicaixia, 2019-01-16:
We commented anything related to right-leg. Now the icub can start calibration, but still print out many errors, given in the log file.  Not sure if those errors will have other problems.
[yarprunlog_17_01_2019_00_10_07.log](https://github.com/robotology/QA/files/2765034/yarprunlog_17_01_2019_00_10_07.log)

julijenv, 2019-01-16:
here ids the cable that could have been snapped or broken.  labelled - 9N1
[image]
[image]
[image]
let us know

pattacini, 2019-01-16:
This is clearly a robot maintenance related issue, thus I'll move it to `icub-support`.